meerk40t/meerk40t #288: Export SVG type might mishandle text objects.

tatarize, 2021-01-07:
Text objects are likely poorly saved, both in position and in font.

gregk40, 2021-01-13:
Same problem, with a design created with inkscape, do not open correctly with meerk40t. works fine with whisperer or gimp, so .svg is correct.
Font is not respected, nor scale/placement of text, and visible layer is not respected
am on linux ubuntu, inkscape is 0.92, meerkat is 0.6.14

tatarize, 2021-01-14:
That's mangling on import. Do Text to Path before sending it to MeerK40t. It's the standard work around currently.

gregk40, 2021-01-18:
> That's mangling on import. Do Text to Path before sending it to MeerK40t. It's the standard work around currently.

It works, thank for help!

inspectionsbybob, 2021-01-20:
Just checked latest beta 06.15.Beta5 and it is still rendering weird. 

Worse, is if you attempt to correct the layout and SAVE it, the file is corrupted to the point that InkScape can't even display it.

tatarize, 2021-01-20:
The work-around kinda worked. In reality it's still quite broken.

tatarize, 2021-02-10:
This might be heavily remedied in 0.6.19-beta2.

#347 Duplicate.

closed 2021-02-11 by tatarize