meerk40t/meerk40t #293: Please walk through Using the Camera to engrave an image on an object

inspectionsbybob, 2021-01-16:
My Camera Came, and it is Firmly mounted on the Lid
I have the 6x9 matrix printed out
I have a Small "Heart" Inkscape file I would like to engrave on a test piece of material.

Please walk me through Aligning the Camera and the Work bed and then Engraving an object with an image.  (or point me to a YouTube video)

When I select the image, and tell the laser to go to the center of the image it goes WAY of grid, so I know something is set/Setup wrong.

My "Work Area Platform" is 300mm x 200mm, and I have set "home" to be the upper left corner of that.

This is the Camera I am using:
https://www.amazon.com/dp/B08MTBYGP5/ref=cm_sw_em_r_mt_dp_CeLaGbS73C9QP?_encoding=UTF8&psc=1

0.6.15Beta 4 recognizes it immediately and it sees the grid, and lets me drag the corners to the corners of the grid.  There is minimal distortion. 
(I reprinted the grid with a border so I can find the white on white corners)

The new camera
[image]

A piece of test material on the sacrificial mesh bed
[image]

The Laptop Hub for the Laser, the camera, and my traveling chicklet.
I got a 6" USB A>B cable which makes everything neat and tidy.  The white cable goes to the camera, the short black (left) cable goes to the laser, and the right black cable goes to the Laptop running MeerK40T.
[image]

I am still trying to learn Meerk40T as a user, so write the instructions as if you are talking a new user through setting up the first time... you can be terse, but don't assume too much...  

I am currently running Beta 4, I would try .70 but don't see anyplace to download it...

joerlane, 2021-01-16:
It would help to see an image of what your camera is looking at. I have mine set to see the bed properly with the lid propped partially open; and also view the job fully with the lid closed. That may or may not be possible or necessary depending on the specific mounting position of the camera, and the field of view of that lens.   

Anyway. The basic procedure is to open the camera view, and begin by using the test grid in different places to correct any distortion given the way the camera is positioned. The button for that correction is in the upper right. You will need to do this a few time until lines for the bed appear straight or diagonal, and not rounded or curved. The correct Fisheye checkbox will allow you to see the difference between corrected and uncorrected view to see if you need to do more checkerboard corrections. Leave the fisheye correction on when you're done.

Aforementioned fisheye correction will correct for lens aberrations of standard webcam lenses, as well as extra wide fisheye lenses (something that may help depending on what your camera view looks like). I happen to have a stick on fisheye lens designed for cell phone cameras in front of my lid mounted webcam, which allows me see the whole job while it's running with the lid closed.

You mentioned that your bed is "set" to 300x200; but is that it's actual usable size? You need to either set the bed size in preferences to be the correct size, or use a smaller size if you want to mark it off for something like A4 or 8.5"x11" paper or the smaller honeycomb you have installed. You could measure your honeycomb and use that instead if you wish. Notice when you set bed size in Preferences it will change the bed size shown in the MeerK40t main window. It also may help to set the Shift Home Position in Preferences to be at the exact upper left corner of your honeycomb (or where ever you want to index your material).

Finally; adjust the perspective handles in the camera window to the outside of whatever "bed" you size you chose. Be that your honeycomb, a sheet of paper, or the full dimensions of the bed. You may need to expand the Camera window larger to find those perspective handles; or zoom the camera view.  If using the full dimensions you can mark it off by engraving a rectangle on scrap cardboard as a guide; or using the laser test button to mark the corners at the extreme edges (move the head with Navigation controls a known distance, then pulse laser to mark the point on test material). Once set; check the correct perspective checkbox and leave it checked. Lastly, send that image to the bed/scene main window in MeerK40t with the Camera button in the upper left of the camera window. 

Now when you load your file, you can place it with reasonable accuracy using the camera view that was sent to the main window. Hint; you may need to send a new snapshot after loading your test material so you can line it up visually with actual working stock. That is the point of using the camera view after all.

inspectionsbybob, 2021-01-16:
OK, several tests, changes and so on and I got it CLOSE!
It was not easy to follow the YouTube video, but it got me going in the right direction.

The Camera and the laser DO NOT ALIGN !

This was what was on screen, I centered my Heart image on the test material
[image]

This is what printed
[image]

Off by quite a bit.... enough that if the target was smaller it might miss....

How do I alight it so what I see and what I print using the camera are the same??

joerlane, 2021-01-16:
Looks like your ruler is still set to a bed width of 300; but your perspective is NOT adjusted to the dimensions as bed size preferences??

inspectionsbybob, 2021-01-16:
Ah... that was what I was wondering, do the dimensions of the drag dots need to match the physical dimensions??  so, if the bed is actually 270 x 190, you enter that as the bed size in preferences?  Not sure i follow the setup but will try that in the morning.

Need a note somewhere that to "calibrate" the camera to your bed  you need to:

1. Set 'Home' to the corner of your work space.
2. perspective/fisheye correct
3. the dimensions of your 6x9 grid box must match your "preferences" work size??
4. anything else so that things line up??

joerlane, 2021-01-16:
Yes. You'll notice when it sends the image to the bed; that it only sends the cropped part which fills the whole bed. The rulers in MeerK40t still show the full size; so it's a completely different perspective than what you attempted to mark. 

You could leave it at 300x200 if you adjust your perspective handles to be at those limits of the bed (that's how mine is setup)... But in your case, those limits would be outside the honeycomb; or even the outer frame of it. 

You'll also need to use shift home position in preferences to use the view perspective you posted earlier, since the home location needs to be in the upper right corner of the image as marked off by your perspective handles.

joerlane, 2021-01-16:
The grid only corrects distortion. It is only used for that purpose and has nothing to do with your physical bed size setting.

The cropped perspective view does need to match the bed size; otherwise you'll be off by the difference. Refer to the rulers if in doubt. Does the background image from camera view match those rulers?

inspectionsbybob, 2021-01-16:
Ah... watching the Youtube that wasn't clear as he kept playing with the perspective box to zoom... which I think was misleading...

It sounds like the perspective box should be set and not touched unless your camera is moved. I assume it keeps the perspective settings between sessions unless reset??

I have lots to do on the K40 this weekend, unfortunately not much is actually printing stuff... Mostly cleaning up wiring, fixing things that are jurryriged and fixing an exhaust leak on the vent?

joerlane, 2021-01-16:
Yes. It will save settings for perspective between sessions. So you should only have to do the full setup once. As long as the lid returns to the same spot reliably for camera view; and the bed doesn't move around.

inspectionsbybob, 2021-01-16:
ok, reading the thread over again I got confused

The perspective correction area can be larger than your work area ?

What two values need to match so that drag-n-drop works?

The first is that the upper left corners must match...

what's the next location or dimension has to match?

joerlane, 2021-01-16:
Yes. Upper left point has to match. The home position, or shifted home position needs to be set to the upper left corner of camera perspective crop. 

The other point is Lower Right, which needs to match your Bed Size in Preferences. So if your lower right point is less; like 250mm x 180mm; you would set the bed size to that, and adjust camera crop/perspective to that point on your bed. 

You'll set upper right, and lower left as well to complete the square. You can use Navigation to move to each corner and then fire the laser to mark the positions. When the bed size matches the image you cropped from the camera, it will be drag and drop.

tatarize, 2021-01-16:
Also, if you want your corners:
`move_absolute 0 0`
`move_absolute 100% 0`
`move_absolute 100% 100%`
`move_absolute 0 100%`

Put like a post-it note there and fire the laser. Then you'll have the points that are logged as your corners and put the perspective corners there and it should line up.

I woulda answered this earlier but I was working on camera, I thought I might be able to just slog through and make a wizard but I burnt a lot of time with some code that worked so badly I replaced it with what I started. Then I added in multi-camera and multi-setting support in 0.7.0. I hope to add in some wizard stuff for camera rather soon.

The plan is 3 stages:
1) Define your bed size. (You don't need it perfect but you need to define the physical size to camera view size)
2) Correct any fisheye distortion. You should get the ability to make the 6x9 checkerboards from that stage in whatever size you want. These should be put on something stiff so they don't move weirdly.
3) Mark the corner. Go to each corner or whatever you defined as a corner and set the perspective markers there.

In theory that should do it. You should be pretty close to spot on after those steps. The only somewhat problematic issue might be the bed size. Since it's defined globally but would in theory be fine for you to use a smaller size than your actual bed size. I think maybe it should just define the size of the area your camera will be tracking and maybe go ahead and position that within the bed. Might require some more thought.

inspectionsbybob, 2021-01-16:
Ok, after finally get the outside vent fixed I got to play with Meerk40T

I re-re-re-Homed the bed and dimensions....
It is really close, but the rendering on screen and the engraving on the material don't -quite- match, in this case it may be the Rendering As  pure images that are "raster wizard"-ed seem to be a bit closer to where they are supposed to go.

I was able to demonstrate the process this morning.
(The K40 Had already been calibrated)

- Open the Lid and place a Material Sample on the Bed
- Take a picture of the Bed with the material on it.
- Open an Image, re-scale the image to fit on the Sample
- Raster-Wizard the image
- Drag the Rasetered image onto the sample where it is wanted
- Close lid and Hit "Start Job"
- 
Done...

It worked that way several times.  The placement is off just a touch, so I have to double check my  preferences dimensions.
(I am going to do a quick video to make my **GlowForge** Friends Jealous with the camera)

It MIGHT be better to have a sample drawing, and to print it and then measure the dimensions carefully...
Don't know, I know the calibration is very very key to the Camera App working the way people expect it to.  

My Mesh bed is smaller than my Max work area, But I have my 6x9 grid printed that fits over it, and I measured the corners carefully on that... I may be off 1mm in 11" which isn't bad.  I have a Positioning Dot on my head, so I can place the material in the same place each time.

I am working on coming up with a procedure that is repeatable each time, both with and without having to calibrate the bed...

I edited the 6x9 grid to have a box around it so you can find/locate the white on white corners and carefully measured the dimensions so I can feed them to preferences once I have "Calibrated" with the Perspective crosshairs  

[image]

tatarize, 2021-01-17:
That might actually fail a bit more. My understanding is openCV is looking for the intersections with the black and white between the pattern. That's why there's 6x9 when it's 7x10 boardwise. That's intended to give a grid of obvious intersections that can be determined and then checked for fisheye and corrected. If this paper was a flat uniform grid, what would the fisheye be to correct that.

Also, it doesn't care what size the actual checkerboard is itself, it cares how large the bed is. We are moving the image from the camera to the image to the bed. And that bed image has a specific actual physical size and the picture taken is made to be that actual physical size. Consequently the outlined edge of the image needs to be a picture of the same physical size as the actual physical size we well end up using.

In lightburn it does this by actually burning a specific image, and since it was burned its actual size is by definition known.

inspectionsbybob, 2021-01-18:
I am going to close this as it is "working" and the thread above gives a LOT of the needed info, but all this needs to be re-worked and put into a Help/Wiki page,

closed 2021-01-18 by inspectionsbybob