meerk40t/meerk40t #300: Error when Trying to execute, useful but unfriendly error msg.

inspectionsbybob, 2021-01-19:
Imported an image
Raster Wizard'd it
Added Basic commands
Edited the speed and power
Tried to print and got this....
From what I could figure out I was trying to Engrave print a raster Photo, it wanted to Raster print the photo....

[image]

Got it to work, but was surprised at the user-unfriendly error message... Useful to a coder/hacker, not to a commercial user.

Not sure the difference between "engrave" and "raster" when it comes to a Photo that has been Raster Wizard'd...

tatarize, 2021-01-19:
The error there isn't intended to be user friendly. It only exists because I missed a bug.  Lemme try to duplicate.

tatarize, 2021-01-19:
You should put the image in an image operation, putting in it in a engrave is supposed to just draw the outline of it. That feature broke however. I'm correcting that. But, you still need a image operation for the image. You could do a raster but since it's already an image it should do some passthrough stuff.

tatarize, 2021-01-19:
https://github.com/meerk40t/meerk40t/commit/3e7d5e1c3080155dc9fed4bd7d5c6edc431302b3

It's useful to me, it tells me exactly what the problem is, where it is, what line and why it failed.

Engrave is a vector operation it does a bunch of straight lines that map to shapes. Raster is a series of pixels that can be traced in various directions. Image is a raster operation where each image is done in series rather than in parallel and all the step size is passed through.

tatarize, 2021-01-19:
Also, changed the message a bit to make it clear the error isn't for the end user, but the development team.

```python
        dlg = wx.MessageDialog(None, err_msg, _('Error encountered'), wx.OK | wx.ICON_ERROR)
        dlg = wx.MessageDialog(None, err_msg, _('Please Report to Development Team'), wx.OK | wx.ICON_ERROR)
```

It now is still not friendly, but more helpful in that it tells you what to do with that message. It also makes a text file of the crash in the working directory.

inspectionsbybob, 2021-01-19:
> 3e7d5e1

If at all possible, error msgs need to have an END-USER helpful error msg
included, either at the beginning or the end.

Long Ago, back in the days of Pascal & 'C'  we wrote all our error messages
with a conditional in front of them to determine what LEVEL of DETAIL to
display.
When we issued a full release, one of the flags we gave the compiler was to
turn off debug.
Our error messages STARTED with an End-User type error message, and then
depending on the Debug flag we would display more and more information.  We
had a "hidden" option for debugging where we could tell an end-user to do a
particular sequence and Debug would be enabled.  That way, even on
production code, when we had a weird error we couldn't reproduce in house,
we could have the end-user turn it on and then send us the logs/error msg.
Bob

tatarize, 2021-01-21:
I don't have any access to the crash. If I knew it was going to happen, I would have prevent it from happening. When they happen I tell the user, and write a log, and ask that the end-user tells me is the only real utility the crash has. Most of the crashes there are internal python crashes. I could catch them. I could try to recover from them. But, it's best to know where they are and correct them when found.

tatarize, 2021-01-21:
I mean it's *entirely* possible there's a better way of doing that. Maybe my current method of hoping they send me the error is a problem. I should actually ask them to send the error log to me by some magical server voodoo where I can get text files sent to the meerk40t group.

"MeerK40t encountered and error. Good news end user. You now have the ability to help meerk40t's development by sending us the bug. Push this big friendly button with some internet access and I'll send this text file here to the very nice and friendly people in a cloud."

Which would likely be great to do, but I dunno any service that could do such a thing. Though the program itself could entirely permit server connections as such.

I suppose I could maybe have a little rest server in the actual web or maybe some already established free for open source service would allow such a thing. Not sure the specifics but it could be helpful to both etc.

inspectionsbybob, 2021-01-21:
There are enough programs that track and report our movements, having a
program report errors wouldn't be too much of a stretch.  Don't know if i
would leave it active in production code, but it seems reasonable to have
that part of BETA code "This programs is BETA and as such has additional
code that will report errors back to Meerk40T. No personal information is
collected or reported, only error codes and conditions"

It shouldn't be that hard to have the system put together a packet and send
it to the standard mail handler to be sent out.

something like:
https://stackoverflow.com/questions/27874102/executing-shell-mail-command-using-python

you would need to have a dedicated email setup.  You could probably do the
the same thing with FTP or other protocols, but email should be easy.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-01-21:
Uses an OS callback that doesn't exist in windows. I guess I could do something with meerk40t.com to send tiny files to a server if there was need for that. Such things would make tracking down bugs a lot easier. I'm sure plenty of people hit crash bugs and don't tell me about them. Giving them a real thing to do would certainly be kinda user friendly. Though I would more prefer a general solution that others here could see and access. Not that a send this file somewhere button would be too weird. Maybe some pastebin sort of thing has anonymous protocols to let you do that and have it searchable.

inspectionsbybob, 2021-01-21:
do you want me to find a script that will work in win10? I thought Meerk40t
was writen in Python?

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-01-22:
I'm coding something up maybe. I'll see if it works for 0.7.0 which would basically open a socket and try to upload it to a server. Could likely set something up pretty easy. And might be a lot friendlier but also a lot more useful.

tatarize, 2021-01-22:
It is written in Python. The problem there is it's making an OS call to send the email. That OS isn't guaranteed. We actually support Windows, Mac OSX and Raspberry Pi. And totally should support some flavors of Linux other than the one on rPi. The problem with that code is it won't work on all three OSes. I tried to code something in pure python, still messing up the http headers but it's pretty closer.

tatarize, 2021-01-22:
[image]

I got it uploading the crash logs to a throw away-ish server. There's likely other services too or I could set others up to accept them.

inspectionsbybob, 2021-01-22:
looks good to me!
That -should- help a bit in the long run...

The only ones it won't help with are the ones that take down the OS as
well, but hopefully those are few...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-01-22:
Gmail is a good place for such responses as well.

Just looked at GitHub, looks better with statuses....

A suggestion is in those emails when it "phones home"  is to put the
VERSION and OS in the message.

The VERSION should also appear in/on the ABOUT page.  There is nowhere in
the software what the VERSION is... It is useful to be able to determine
what version you are running.

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

tatarize, 2021-01-22:
I already had the version since it really is sending the file. However, it didn't say it there. It is corrected though. I removed the part about personal information since your regular username could be in the User Directory of the error log. And then I changed it to say "This is all we will send." but that wasn't quite true either since I send the log and the log had the version in it.

closed 2021-01-30 by tatarize