meerk40t/meerk40t #302: Terminal needs SOME documentation

inspectionsbybob, 2021-01-19:
A great deal of power is in the direct execution of command through the terminal, but they aren't documented.
 
Some things I have "figured out" from emails like

move_absolute 0 0
move_absolute 100% 0
move_absolute 100% 100%
move_absolute 0 100%

But I am sure there are LOTS of other commands, but I have no idea where to look.  A Side/bar as an OPTION in terminal mode with command syntax would really be helpful.

What fires the laser for how long, what TOGGLES the laser
What sets the Travel speed
Lock/Unlock the rail
Are there any controller variables that are "remembered" that would be useful to know about?

Here is a Wiki I found of commands (http://edutechwiki.unige.ch/en/Lhystudios_M2_Nano#Commands) It includes some Meerk40T references

tatarize, 2021-01-20:
>What fires the laser for how long,

`pulse <duration-ms>`

So console: `pulse 200` fires the laser without moving it for 200 ms.

> what TOGGLES the laser

`+laser` (WARNING: DO NOT JUST TYPE THIS!)

This would typically be bound to a key: `bind space +laser` when something starting with a + is bound it's triggered on keydown and the `-` form is triggered on keyup. So here hitting your spacebar fires your laser in place. (WARNING: DO NOT ALLOW A CAT TO BURN DOWN YOUR HOUSE!)

>What sets the Travel speed

`speed <speed-mm/s>` in theory when the laser is rastering you can type: `speed 200` and the laser will start moving at 200mm/s. Or `speed 50` and the laser will change speed to that speed. You can also do `speed +10` which will go 10mm faster than the current speed. And `speed +10%` which would make it go faster by 10% from the previous speed. Keep in mind these speeds would be lost when the laser changes its speed in common use. So the utility there is slightly mooted.

> Lock/Unlock the rail

* `lock`
* `unlock`

These have no arguments.

> Are there any controller variables that are "remembered" that would be useful to know about?

Within the actual controller itself? I don't think so. But, they are as follows.
* bool: quit -- Flags the controller to shutdown when the current processing data is finsihed. This is so the CLI can do `-a` flag and have the controller thread terminate and allow it to exit the program.
* int: usb_index -- This is the number of the usb. So setting this to 0 only gets the 1st device. 1 the 2nd, 2 the 3rd.
* int: usb_bus -- This restricts the controller to attaching to only this usb bus index.
* int: usb_address - This restricts the controller to attaching to only this usb address.
* int: usb_serial - This does nothing.
* int: usb_version - This restricts the controller to attaching to a CH341 chip only of this exact version. This can be used to tell one board from another board if you have more than one.
* bool: mock -- This pretends you have a usb device while not really having one for debug purposes.
* int: packet_count -- This is the count of the sent packets by the controller.
* int: rejected_count -- This is the count of the rejected packets that failed. This value is useful but displayed in Controller and can show you if your connection is sketchy.
* bool: autolock -- This is in preferences and locks the rail when finished with a process.
* bool: autohome -- This adds a home in the spooler it doesn't do anything in controller.
* bool: autobeep -- This adds a beep command in the spooler it doesn't do anything in controller. 
* string: board -- This is the board type. M2 is default but, A, B, B1, B2, M, M1 are also allowed. Setting this will mess with your speed codes and make the machine go really slow or wrong if setup wrong.
* bool: rotary -- This is whether your rotary is set to work. Does nothing in controller.
* float: scale_x -- This is the scale in the x direction for your rotary. It does nothing here.
* float: scale_y -- This is the scale in the y direction for your rotary. It does nothing here.

Version 0.7.0 is adding bool: strict, this tells the controller to only ever engage with RB (Bottom, Right) directions initializations on the believe that versions of the M2 board prior to revision 4 circa 2014 which have a different stepper chip cannot actually correctly deal with other directions.  We lack the hardware to verify this, but we think it's the result of #272, which was a bit of a shock to us.

That wiki of commands are the LHYMicro-GL commands found within the board and reverse engineered. I know, I wrote that wiki.

tatarize, 2021-01-20:
I could actually expose `fix_speed` found in the LaserSpeed code. Which makes all speeds 8.9% faster which actually makes them accurate. For an unknown reason the chinese software numbers were uniformly off by this amount. Consequently when I wrote LaserSpeed (which predates MeerK40t), I perfectly duplicated the values there because I needed to get Scorch to use the code for Whisperer. And not wanting to break things those were left in place. But, a speed of 100mm/s is actually going to go 91.1mm/s. The transitions between different accelerations was bound to that and I wanted people to be able to keep their whisperer speeds. So it was left there.

In fact, I've now corrected these in 0.7.0
https://github.com/meerk40t/meerk40t/commit/dd7edb92a6904b64bdbc921a952f8381250d5cbd

tatarize, 2021-01-20:
0.7.0 is getting a more advanced `help` command which exists in a more primitive form in 0.6.x

```
--- Context('1') Commands ---
left            <direction> <amount>
right           <direction> <amount>
up              <direction> <amount>
down            <direction> <amount>
move            move <x> <y>: move to position.
move_absolute   move <x> <y>: move to position.
move_relative   move_relative <dx> <dy>
home            home the laser
unlock          unlock the rail
lock            lock the rail
egv_import      Lhystudios Engrave Buffer Import. egv_import <egv_file>
egv_export      Lhystudios Engrave Buffer Export. egv_export <egv_file>
egv             Lhystudios Engrave Code Sender. egv <lhymicro-gl>
usb_connect     Connect USB
usb_disconnect  Disconnect USB
start           Start Pipe to Controller
pause           Pause Controller
resume          Resume Controller
abort           Abort Job
pulse           pulse <time>: Pulse the laser in place.
speed           Set Speed in Interpreter.
power           Set Interpreter Power
acceleration    Set Interpreter Acceleration [1-4]
--- Global Commands ---
bind            bind <key> <console command>
alias           alias <alias> <console commands[;console command]*>
consoleserver   starts a console_server on port 23 (telnet)
grid            grid <columns> <rows> <x_distance> <y_distance>
element         element <command>*: <#>, merge, subpath, copy, delete, *, ~, !
path            path <svg path>
circle          circle <x> <y> <r> or circle <r>
ellipse         ellipse <cx> <cy> <rx> <ry>
rect            adds rectangle to scene
text            text <text>
polygon         polygon (<point>, <point>)*
polyline        polyline (<point>, <point>)*
stroke          stroke <svg color>
fill            fill <svg color>
rotate          rotate <angle>
scale           scale <scale> [<scale-y>]?
translate       translate <tx> <ty>
rotate_to       rotate_to <angle>
scale_to        scale_to <scale> [<scale-y>]?
translate_to    translate_to <tx> <ty>
resize          resize <x-pos> <y-pos> <width> <height>
matrix          matrix <sx> <kx> <sy> <ky> <tx> <ty>
reset           reset affine transformations
reify           reify affine transformations
operation       operation <commands>: <#>, *, ~, !, delete, copy
classify        classify elements into operations
declassify      declassify selected elements
note            note <note>
cut             group current elements into cut operation
engrave         group current elements into engrave operation
raster          group current elements into raster operation
step            step <raster-step-size>
trace_hull      trace the convex hull of current elements
trace_quick     quick trace the bounding box of current elements
grblserver      activate the grblserver.
image           image <operation>
halftone        image halftone <diameter> <scale> <angle>
optimize        optimize <type>
embroider       embroider <angle> <distance>
plan.*          plan<?> <command>
ruidaserver     activate the ruidaserver.
camera\d*       camera commands and modifiers.
start           Start Camera.
stop            Stop Camera
fisheye         fisheye (capture|reset)
perspective     perspective (set <#> <value>|reset)
background      set background image
export          export camera image
camwin          camwin <index>: Open camera window at index
loop            loop <command>
end             end <commmand>
timer.*         timer<?> <duration> <iterations>
register        register
context         context
set             set [<key> <value>]
control         control [<executive>]
module          module [(open|close) <module_name>]
modifier        modifier [(open|close) <module_name>]
schedule        schedule
channel         channel [(open|close|save) <channel_name>]
device          device [<value>]
flush           flush
shutdown        shutdown
rotaryview      Rotary View of Scene
rotaryscale     Rotary Scale selected elements
window          wxMeerK40 window information
refresh         wxMeerK40 refresh
```

tatarize, 2021-01-20:
In addition, these commands also will return the doc_string of the command permitting much more advanced help to be provided with `help <command>`

Some of this might change a bit when I have commands define input and output types as string variables. Some commands like `camera.*` actually just give a camera type and the sub-commands within that take the camera data_type and modifies it, allowing command chaining in 0.7.x.

```
camera\d*       camera commands and modifiers.
start           Start Camera.
stop            Stop Camera
fisheye         fisheye (capture|reset)
perspective     perspective (set <#> <value>|reset)
background      set background image
export          export camera image
```

These are all part of the camera command. So `camera stop start` reconnects the camera calling stop and start on the camera. Whereas `camera0 background` does sets the bed background to the last frame of camera 0. Which is how multiple cameras work in 0.7.0, in theory this could also allow, export to export an image which could then be modified in image commands which are currently set in paradigm.

tatarize, 2021-08-12:
Did some documentation on this to help settle this issue:

https://github.com/meerk40t/meerk40t/wiki/Doc:-8.-Console-Terminal

tatarize, 2021-09-25:
Documentation is pretty good did a 20 minute video on it. 

https://www.youtube.com/watch?v=c_QBZlNvhVo

closed 2021-09-25 by tatarize