meerk40t/meerk40t #304: Why No "Project Save" or way to save Operations Clasifications

inspectionsbybob, 2021-01-19:
There is an "Open Project" but no equivalent "SAVE".

If there were, would that be where "Operations" classifications (color, speed, power) might be stored?

Just checked, The Operations definitions are saved between open/closes, but NEW wipes everything and "Set Basic Classifications" doesn't remember either.  (Is the list of "Basic Classifications" stored somewhere editable?)

tatarize, 2021-01-20:
Short answer. These save in save project. They are stored in the SVG as operations and reload when you load the meerk40t svg file.

---

This is accurate. The reason is generally bad-ish design. There is an overarching problem here that you've sort of pegged. This could be collated into a single issue outlining the problems. Mostly these have a lot of different issues that ultimately need to be addressed probably with one good solid solution.

#289 "Operations saved/loaded/defaulted gets confusing"
#256 "Cutplanning within a closed shape" - Also touches on some of this, later in the response.
#209 "Project save reclassifies rather than storing items. " Is a noteworthy problem within this same scheme.
#239 Materials Database. Could also be classed in here.

There's not a coherent paradigm with regard to operations or specifications of how they should work. They define processes and elements within them get compiled into cutcode which is a sort of generic way to express laser operations. But, the dealing with them is rather slapdash.

* There are two predefined static fixed values. Set Basic Classification and Other/Red/Blue. These are not saved. These are hard-coded.
* Operations you are using are saved on close. This applies to the operations themselves, not to anything else within your current work.
* Operations are saved with svg files you save and stored in the SVG itself.
* When operations saved in an SVG file are reloaded these overwrite whatever your current operations were.
* This load mechanism holds true even loading blank SVGs. This permits you load an operation for a specific material if you wanted to save custom empty svgs with just operations inside them.
* Materials databases are a very reasonable feature which closely parallels the point you're making here. It is entirely reasonable to want some particular custom and settings for certain materials, and have these saved to be reloaded at your leisure.
* RDWorks defines colors with operations in a toolbar at the bottom. Pressing the button, changes the strokes of the current element which makes it part of that operations. Operations and elements have a strict 1:1 relationship. If you want the same path in two operations, you would make a second copy of it.
* Whisperer has predefined static operations where red = cut, blue = engrave, everything else = raster. This remains uniform and unchanging. Your settings for these are readily visible and you simply hit the button to execute that operation.
* LaserWeb has a materials database that lets you define parameters and speeds for particular materials and load them up. Visicut is similarly inclined. 

[image]

But, having your own settings that are different from the current settings and saving them and reloading them would an improvement. But, the correct synthesis of all these problems into a proper paradigm solution doesn't exist yet.

Also, the project-save is kinda a misname since it's not really a project. There's not a concrete start and end for your work. If you load additional svgs these don't get a different tab to work on there (notebook functionality is possible with aui which is now partially part of 0.7.0). There's just all the things that currently exist and vague slapdash loading and changing of things in non-optimal ways.

tatarize, 2022-07-20:
There are some cursory methods of material database. These more advanced settings are expected to be added but the classifications do currently save correctly etc.

closed 2022-07-20 by tatarize