meerk40t/meerk40t #305: Please close "closed" issues

inspectionsbybob, 2021-01-19:
Looking at the log, I can't tell what is still open.  Anything that has been fixed, should be closed for consistency and readability.  (and the project looks better too)

tatarize, 2021-01-20:
This is usually done when the new version goes live. I usually go through and mark things as fixed in dev or fixed in main. And close them at that point. But, while things are being corrected still or ideas are posted, even duplicate ones, I tend to leave them be.

tatarize, 2021-01-20:
I do tend to preserve some issues because they contain ideas or suggestions that I don't really want to lose. And some of them should be renamed and labeled better. Usually though we close a bunch of stuff when it's corrected in the main branch and just sort of flag it with labels while it's worked on. Especially if hyper-specific like a bug or whatnot.

tatarize, 2021-01-22:
Okay, I added useful and "sane labels" to the issues here. I then applied them to all the open issues here. And let you have some issue control. There are certainly some of them that can be closed. Especially those that were in the very recently published 0.6.16 (and 0.6.15 (nuked)).

closed 2021-02-11 by tatarize