meerk40t/meerk40t #308: Dark Theme , OSX

rafaljot, 2021-01-21:
Dark Thame is nice but there are some minor color issues like.

1. Black on dark grey
2. White on yellow
3. White on light green

[image]

[image]

[image]

[image]

tatarize, 2021-01-22:
For the icons. I am pretty sure I could flip their color schemes and invert them in places. But, I don't really know when dark mode is on or not. I might be able to set them to the same color as the text depending on dark mode works. The best correction for the console color issues might be to not use angry fruit salad colors to begin with. Maybe colors that are more neutral between light and dark but still sort of reflect the current state of that thing?

tatarize, 2021-01-22:
That icon thing in device manager is just a bug. Those are *supposed* to be up and down icons that are disabled, but apparently they just move the corner and overlap for unknown reasons. I don't really allow reordering there. But, rather than disable they get drawn weird.

tatarize, 2021-01-22:
Frog figured out: wx.SystemSettings().GetColour(wx.SYS_COLOUR_WINDOW)

Totally gives whether dark mode is on. So some specific things could be changed based on whether dark mode exists or not.

tatarize, 2021-01-22:
https://dzone.com/articles/wxpython-creating-dark-mode

Dunno, but it might be possible to set the background color for various windows and get it to work. Some particular widgets might not work.

----

Like two of the images you posted are just bugs. If there's something design wise, that should be shifted I'll need specifics. Gui design is not my forte.

tatarize, 2021-01-30:
I fixed the couple bugs but I'm not a gui design person so the remaining bad darktheme stuff isn't very apparent what's required there.

tatarize, 2021-01-31:
@rafaljot -- These may actually be fixed. I hacked the icons to invert, and I force set the color to black in the cases where I have a giant colored buttoned. Removed those two clearish bugs and should be in the beta https://github.com/meerk40t/meerk40t/releases/tag/0.6.18-beta1 I'll ping @joerlane about pumping out a beta OSX version.

RuKeBo, 2021-02-01:
Can you access Dark mode in Windows? I can't find a setting for it.

tatarize, 2021-02-02:
No. No dark mode in Windows. There's some idea one could be hacked in there. Just run some code on each window on start to recolor such things but currently none of that. It's an OSX thing apparently.

tatarize, 2021-02-08:
@rafaljot Please verify 0.6.18 corrects these issues for you. Frogmaster should have also performed some additional tweaks that should work for Dark Mode.

closed 2021-02-11 by tatarize