meerk40t/meerk40t #311: Microsoft flagging MeerK40T as Dangerous?

inspectionsbybob, 2021-01-22:
I have had this come up each time I download a new release.

Not sure if it is GitHub or MeerK40T

[image]

I reported it as safe, you may want to do the same... or fuss as GitHub

tatarize, 2021-01-22:
I very specifically check the thing with virustotal explicitly to avoid false positives.  Though, I think that might actually just be because it's an exe. And I think explorer might block them all. As well they should.

https://www.virustotal.com/gui/file/dbf63e692458c9a38b24c6971a5d71583d351c8dcd713762261b3e7dfee76a34/detection

Microsoft actual antivirus does not detect anything. I downloaded it with Chrome and nothing reacted to this.

tatarize, 2021-01-22:
SecureAge APEX: Malicious
Cylance: Unsafe

These are the vaguest of the vauges. My general rule of thumb is if microsoft and the major virus things pass, we're good. 

As always you can check the source code and run it from source if you feel unsafe. You cannot actually prove a binary is what it says it is. Also, with 0.7.0 this will become a lot more popular since `pip install meerk40t[gui,dxf,cam]` is pretty easy.

Sophist-UK, 2021-05-15:
Microsoft Security does not flag 0.6.22 but does flag 0.7.0b23:
```
Threat detected: Backdoor:Win32/Bladabindi!ml
Alert level: Severe
Category: Backdoor
Details: This program provides remote access to the computer it is installed on.
```
[Learn more](https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Backdoor%3aWin32%2fBladabindi!ml&threatid=2147748148)

tatarize, 2021-05-15:
They are compiled differently. I compiled 0.7.0b23 to try to get around the dll issue but I didn't keep compiling the bootloaders to avoid all the false positive. I was going hit it a bit harder after getting the devices working.

Sophist-UK, 2021-05-15:
It's a beta - so don't give this too much priority cf. functionality.

closed 2021-08-12 by tatarize