meerk40t/meerk40t #312: 6.15/6 Broke Home?

inspectionsbybob, 2021-01-23:
Wanted to start clean, so I re-set everything.

Re-set the perspective, re-set home, re-set dimensions
Went to Home.... Thats -NOT- where I set home.
Re-set Home... HOME.... -NOT- where I set Home...

Double checked dimensions.  Double checked Perspective.  Tried again.
re-re-re-set HOME.... It is NOT going back to where I am setting home as, and that messes up EVERYTHING.

Working on some Power/Speed test charts and Missing the ability to place things using the camera is messing me up.
Was working fine in ~6.15.   6.16 Messed up something and it is staying messed up.

Managed to get K40 controller stuck.  Restarting K40 did nothing.  Restarting Meerk40T did nothing. Got "No USB" message but camera worked fine. Had to Load k40Whisper to reset. Then it cleared the buffer of all the queued commands....  Need a RESET that can get it unstuck and clears garbage out.  Not sure where "HOME" is being stored along with other settings, but there probably needs to be a K40 Folder that can be watched/deleted as needed to do a clean reset.

joerlane, 2021-01-23:
Can you explain how you performed this reset, and re-re-re-reset HOME? I tried to reproduce any of this issue, and was unable; but I really can't be sure what you're describing without more concrete information.

I suspect the "No USB" could have been resolved by one of the top buttons in the Controller window (Hold/Resume Disconnect/Connect). On rare occasions I have to power cycle my K40. But if you say whisperer fixed it; does that mean whisperer could have caused it? Did you have both open at the same time? Or a different version of MeerK40t even?

inspectionsbybob, 2021-01-23:
Everything was behaving until I aborted a run using controller stop.
I tried using the PAUSE button at the top, but the only thing it would let me do after pause was continue. There was no way to abort the job... Everything went south/bad after that.  lost USB communication, lost home, lost my sanity almost...

Need to do a clean re-install. may need to find where "home" is stored and delete that as well for a true CLEAN re-start.

tatarize, 2021-01-23:
After aborting the job. You need to rehome. Abort is a dangerous operation and causes registration loss. It will not know where home actually is because it had to destroy some number of commands in the buffer to kill that job.

inspectionsbybob, 2021-01-23:
i power cycled the k40 and still no USB
I power cycled my laptop, no usb.
I brought up K40whisper, and -IT- reset the controller.

Now I am trying to figure out why HOME is not sticking.

I power cycle the K40, and then use navigation to find my corner, go to settings, and set HOME to current position.

Then if I hit HOME, it should go to to 0,0 and then return to THAT position, and it is not...

The camera is also off... tried doing things in different orders, still not returning to HOME.  

Not sure what happened between 6.15 and 6.16 but something got stuck in somewhere and it is messing me up.

I never have more than one program talking to the K40 at once. I may have Inkscape running so I can go back and forth, but nothing else, my laptop doesn't have enough umph for that...

tatarize, 2021-01-23:
What do you mean why Home is not sticking? 

---

Are you moving the laser by hand or with the navigation before setting home?

inspectionsbybob, 2021-01-23:
Hum, if if Abort can't be used to STOP, we really need a good, safe, reliable STOP button. or a way to stop-stop after pausing.  All my problems started after an Abort when I could see my power levels were way off and I didn't want to waste 5min letting it finish the run.

tatarize, 2021-01-23:
Two things trying to talk to your laser would step on each other's toes. Inkscape does not talk to the laser.

Abort can be used. It just needs a home afterwards.

tatarize, 2021-01-23:
Okay try to set your home position and then look up what that is. Hit F12 to load your Terminal. Type set. This should show all your settings including the home adjustment.

tatarize, 2021-01-23:
```
set
"autobeep" := True
"autohome" := False
"autolock" := True
"autostart" := True
"bed_height" := 210
"bed_width" := 310
"board" := M2
"buffer_limit" := True
"buffer_max" := 900
"current_x" := 0
"current_y" := 0
"device_location" := USB
"device_name" := Lhystudios
"device_version" := 0.0.0
"draw_mode" := 0
"file0" := C:\Users\Tat\Desktop\fgsdfgdfg.svg
"file1" := C:\Users\Tat\Desktop\TP earrings.svg
"file2" := C:\Users\Tat\Desktop\dfgsdfgs.svg
"file3" := C:\Users\Tat\Desktop\PanelMeter14mk1.svg
"file4" := C:\Users\Tat\Desktop\40mmOak21_mk3.svg
"file5" := C:\Users\Tat\Desktop\screenshot.png
"file6" := 
"file7" := 
"file8" := 
"file9" := 
"flip_x" := False
"flip_y" := False
"fps" := 40
"home_adjust_x" := 1182
"home_adjust_y" := 1182
"home_bottom" := False
"home_right" := False
"mock" := False
"name" := 1
"navigate_jog" := 394.0
"navigate_pulse" := 50.0
"opt_jog_minimum" := 127
"opt_jog_mode" := 0
"opt_rapid_between" := True
"packet_count" := 67987
"quit" := False
"rejected_count" := 39
"rotary" := False
"scale_x" := 1.0
"scale_y" := 1.0
"state" := 2
"swap_xy" := False
"uid" := 1
"units_index" := 0
"usb_address" := -1
"usb_bus" := -1
"usb_index" := -1
"usb_serial" := -1
"usb_version" := -1
"window_height" := 586
"window_width" := 1017
```

"home_adjust_x" := 1182
"home_adjust_y" := 1182
is my modified home position.

When I hit home. It homes, then moves to the new location. Then leaves the one little circle exactly where it was in the corner. Is something different happening for you and do you have maybe a video or something I can't figure out what exactly it's doing or not doing and whether it's software related at all.

tatarize, 2021-01-23:
```
> channel open send
Watching Channel: send
> home
b'\x00IPPFFFFFFFFFFFFFFFFFFFFFFFFFFF\xe4'
b'\x00IBzzzz162Rzzzz162S1PFFFFFFFFFF\xc1'
```

The first command there is Home. The second command is adjust the position of the home.

inspectionsbybob, 2021-01-23:
i am power cycling everything.
I am starting Meerk40t 6.16 and doing the following in the following order.
New, clear perspective
put 6x9 grid down
carefully tag corners
verify corner to corner dimensions in windows/preferences
Use Navigation to move to the corner of the 6x9
Set HOME position (preferences)

at that point I should be able to hit HOME on the navigation screen and it should go to 0,0 (absolute) and then go to the new -HOME-.

it is not. It almost seems random where it is going...

inspectionsbybob, 2021-01-23:
getting tiered, need to start fresh in morning after coffee.

too many moving parts, was working on sample current/speed/power test runs when Meerk40t hiccuped

inspectionsbybob, 2021-01-23:
After Coffee I re-did everything.
This is 0.6.16
Paid extra attention to Zeroing and setting size....

I think one of my belts is loose, so I need to research that.... stepping down small movements, it moves and then moves back as if it ran into something soft.  Need to double check my air assist tube isn't messing things up.

I power cycled, made sure the current position was (0,0) and set that as home (set)
I then moved LARGE MOVEMENT forward, and then moved back to find HOME, and then SET it, verified it...

FOR ME....
I think having to do a power cycle to set 0,0, then work backwards to a good corner, and THEN SET, seems to work.
NO STOPS!   Wanted to, but I let the run finish even though It was bad so I would have to go through re-zero-homing.

SUGGESTION:
Need a ZERO button that goes back to PYHSICAL 0,0 and stays there for setting ZERO (0,0) in process to re-set HOME
 (or do better cleanup after an Abort?)

Also, not sure if my carriage is staying put, should I lock it when trying to set HOME?

Zero-ing, and STAYING zeroed is important, and I don't think the current methods are robust enough, ESPECIALLY if you are in an Environment where a STOP may be involved.

closed 2021-01-24 by inspectionsbybob