meerk40t/meerk40t #314: problem with Splines in DFX

gyan007, 2021-01-23:
the splines are not showing correctly in v.0.6.16

[problem_with_splines_in_DFX.zip](https://github.com/meerk40t/meerk40t/files/5859804/problem_with_splines_in_DFX.zip)

tatarize, 2021-01-23:
[image]
[image]

Visicut rotates it and moves it to the bed. But, clearly should be hourglass shaped. It is not.

tatarize, 2021-01-23:
The issue is in ezdxf. I used their code for splines which works great except for rational b-splines which these are, so it used the fallback of using lines.

tatarize, 2021-01-23:
Actually there maybe another approximation method to use first before resorting to this terrible one.

tatarize, 2021-01-23:
Okay, splines will now have 2 fallbacks but, that's fine. Corrected it. But, we finished 0.6.16 and then proceeded to break 0.6.17 instantly with a big can of worms. So lemme go ahead and maybe shove those back in the bottle. I wouldn't go clicking any spinners in GTK based windowing systems, and closing the window that's in the beta... 

Assuming you're a windows user. Would kinda need to know the build to give you with the fix.

joerlane, 2021-01-23:
What operating system are you using MeerK40t v0.6.16 on?

gyan007, 2021-01-23:
I am using windows 10 64-bit and windows8.1 32-bit

tatarize, 2021-01-23:
Okay, I'll have a beta out for you in a minute or two.

tatarize, 2021-01-23:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.17-beta1

tatarize, 2021-01-23:
[image]

gyan007, 2021-01-23:
Cool, Thanks!

gyan007, 2021-01-23:
[image]

It corrected the Big spline but turned small line segments into wave like patterns

tatarize, 2021-01-23:
Hm. Might be the guess it's taking. Really those are also going to be splines but weighted differently and it's taking a solid guess at where they should be.

tatarize, 2021-01-23:
Apparently yeah, higher numbers for better approximations. 5 seems about right for that.

Making a new beta.

tatarize, 2021-01-23:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.17-beta2

tatarize, 2021-01-23:
[image]

I could go higher but the gains seem to be less above this. You could also slightly tweak it to use lines in that part of the path if you're messing with the original dxf. Seems to have a solid approximation vs. cost value.

gyan007, 2021-01-23:
ya, looks good. Using lines for straight paths instead of all splines in dfx would solve the problem for me at least

tatarize, 2021-01-23:
Especially when you're weighting them to make them straight there. It's gotta convert them to beziers so there's a bit of wiggle in the approximation. But, that's solid enough for everything there. At least it's a lot closer and much less noticeable.

gyan007, 2021-01-23:
I  tried importing dxf to Inkscape ->> to svg --> to meerk40 and it came out fine.

tatarize, 2021-01-23:
Inkscape probably has better math for the b-splines. It's likely a better fallback for that. I'm just working with ezdxf. Which is why at first this was marked blocked. Because I didn't think it was doable. But, there's a better fallback there, but certainly isn't perfect. There's likely better algorithms. Especially since you're really making lines there.

gyan007, 2021-01-23:
yeah, Thanks!

tatarize, 2021-01-23:
Also, 5 might be too high, I'm seeing some performance issues. So yeah, either don't really use those or run it through ezdxf. I'll take it under advisement though those approximations are approximate.

tatarize, 2021-01-23:
Oh, nm. I temporarily put that to 10 to see how good it got. Yeah, that's a performance drain.

closed 2021-01-30 by tatarize