meerk40t/meerk40t #315: Error reported at closing

inspectionsbybob, 2021-01-24:
System said to send this to you, so I am...

Saw this when I was closing down for the evening to work on some other things....

[image]

tatarize, 2021-01-24:
I'm guessing that's 0.6.17 beta. Or one of them.

There's a standing issue there with spinners in gtk version doing a strange ordering with the save. It's a kinda trivial bug but a headscratcher on fixing it. 

Mac and Linux users if they have the spinner focused and they close the window. The setting they set in the spinner does not save. So I tried a bunch of stuff to fix that. And then just corrected any crashing it would do. It still does not save. I got rid of spinners for important areas since they have *other* bugs too. Namely the SpinCtrlDouble in GTK only does integers, which is absolute death in a number of places.

I restored the anti-crash stuff after those betas and ran into another windows bug that I thought was maybe caused by that. Loading up navigation and then closing the main window to kill the sub-windows works, but it doesn't shutdown gracefully and often crashes after closing.

It's a known error. I just don't know how to fix it, and might just restore the failed stuff. The idea of maintaining the device reference after window close seems to have opened up a can of worms. And it's fixed in 0.7.0 so it's a bit of a pain in the ass overall.

tatarize, 2021-01-24:
https://github.com/meerk40t/meerk40t/pull/313/commits/9849e065e22ae2a0fe8d10e94bf1c988d7e7c80a

That specific one was just stuck in the last beta. It's fixed for 0.6.17 proper which should be released fairly soon.

closed 2021-01-24 by tatarize