meerk40t/meerk40t #319: drawings get filled once merged and optimized

gyan007, 2021-01-25:
drawings get filled once merged and optimized 
[fill_problem.zip](https://github.com/meerk40t/meerk40t/files/5867978/fill_problem.zip)

tatarize, 2021-01-25:
That's entirely true. Since the difference between two circles as part of the same path and two circles as their own path is that the fill of the two circles as part of the same path might well make a donut shape, whereas two circles otherwise would otherwise just fill in the same area, and part of that area twice.

If you're optimizing it for pure geometry you can select it and hit console and type `fill none` which might be better. The merge and optimize isn't wildly well advertised because it wasn't really perfect. It was good but actually that optimization stuff would get done later as part of cutcode in 0.7.x (or at least that's the plan).

tatarize, 2021-01-25:
I'd recommend unfilling with `fill none` if you're wanting to save that out or whatever. 0.7.0 has a better way of solving that issue.

closed 2021-02-11 by tatarize