meerk40t/meerk40t #321: Crash Report

inspectionsbybob, 2021-01-28:
MeerK40t crash log. Version: 0.6.16
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 1149, in on_right_mouse_down
  File "Widget.py", line 187, in event
  File "Widget.py", line 624, in event
  File "wxMeerK40t.py", line 2285, in create_menu
TypeError: 'NoneType' object is not iterable

I was Playing and I got this....

I had opened an image, and was enlarging it so I could play with Raster Wizard...

inspectionsbybob, 2021-01-28:
I am getting it if I RIGHT click on the image....

Clicking on Raster Wizard at the top seems to work

tatarize, 2021-01-29:
I can't get mine to crash to duplicate that. Does your current run work on 0.6.17? There's some chance it's related to the bug that resulted in 0.6.16 getting nuked.

inspectionsbybob, 2021-01-29:
is 6.17 available yet ?

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-01-29:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.17

Yeah, had to nuke 0.6.16 because of what might be a related bug. And pushed out 0.6.17 really quick.

There's also a buggy beta of 0.7.0 that's two versions in:
https://github.com/meerk40t/meerk40t/releases/

inspectionsbybob, 2021-01-29:
What should I be "playing with" when I get the time...  6.17 or 7.0  ??

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

tatarize, 2021-01-29:
Well first make sure 0.6.17 doesn't have that crash. But, after that some feedback on 0.7.0 might be nice. It has a lot of bugs but they are all more straight forward. Though I'd still run any projects you want in 0.6.17 or double check before running 0.7.0.

closed 2021-02-09 by tatarize