meerk40t/meerk40t #326: please append to log file, not just new every time

inspectionsbybob, 2021-02-01:
recently was arguing with 6.16 and had to exit to recover only to be treated to a desktop FULL of error logs... all of the same thing, all already reported, so I had to go through and (find) delete all the error files...

Please just Open/Append to the log file... not a new one each time, either that or put them in a folder, not the desktop.

tatarize, 2021-02-02:
They are in the working directory. If you put the meerk40t.exe in a folder, all the errors would end up there too.

inspectionsbybob, 2021-02-02:
Hum... acceptable, not ideal.  Would prefer a setting (Windows, settings) that specifically said where to put stuff.

Will close it though

inspectionsbybob, 2021-02-02:
Suggest an upgrade that puts a setting in "Windows, settings" of where all MeerK40T errors and files go by default.

closed 2021-02-02 by inspectionsbybob

tatarize, 2021-02-03:
The problem with that is when I hit an error, I try to avoid touching anything else. Because if it hits a bug in that routine it's a problem on top of a problem.