meerk40t/meerk40t #327: Won't Recognize another USB port

inspectionsbybob, 2021-02-02:
I don't have a dedicated PC to run MeerK40T and the laser, so it gets plugged/unplugged a fair amount.

Today I swapped where my USB Memory stick and the K40 went, and it couldn't find the K40

Swapped the Stick and the K40 back to their "normal" locations and it found it right away, and then proceeded to send all of the "Home" commands I had tried to send when it was on the "other" port.

Don't know if it is a driver thing that you have no control over, or if there is a setting to try different ports

tatarize, 2021-02-02:
You might want to make sure in preferences all your values there with regard to USB are -1 all the way down. There's some need for some versions to disregard the port in some cases. So that you could run a different machine. If that's set in there, then switching the port around could make it fail. Also, I might have to recheck that functionality in 0.7.0, it was changed a bit to work correctly on linux and I should make sure it correctly un-claims but it should if you changed usb ports.

Also, your usb log should have some solidish information about the connection. If you're using the windows default CH341 driver this information will be sharply reduced so if you want fine grain understanding  You could get the Whisperer install .exe and use that to install the libusb driver and then have some finer grained control over seeing what's going on. But, your log should still say what's going on and why it's doing that.

inspectionsbybob, 2021-02-06:
I got an "active USB extension cable" which lets me run to my computer on the other side of the room where the other machines (3d printer and vinyl cutter) are so I won't have to swap ports anymore.  Trying to run multiple K40's could be a problem if it starts "looking" for a machine since the all the nano controllers have the same name and I don't think there is an option of "renaming" them like on smarter controllers. 
I will close this as a conflict with a feature. It assigns a port to a machine, not the other way around.

inspectionsbybob, 2021-02-06:
Fixing this would break "works with multiple K40s" as it assumes a port is a machine. 
Allowing it to "look for a machine" could get the software confused if there were multiple K40s, all with identical USB identifiers, on multiple ports.

closed 2021-02-06 by inspectionsbybob