meerk40t/meerk40t #328: Feature Request: Opacity to factor PPI for Gradients

inspectionsbybob, 2021-02-02:
I was looking at various things to play with and went back to an OLD bug report/comment that "Gradients" don't work well.

Suggestion: Use An objects "Opacity" of 0-100% as a FACTOR against PPI.

If an Object is a "Full" color, ie 100% Opacity, everything would work as normal.   (PPI x 1.00 = PPI)

If an object if semi-transparent, ie 50% Opacity, the existing PPI (from Operations) would be multiplied by .50 whatever it was.

Would recommend another switch/flag/click box to enable/disable the feature.

**[ X ] Use an Object's OPACITY to further control the laser power.**  

Ex: If PPI is set to 650 and an object is 75% Opacity the resulting 
PPI would be 650 x .75 = 487PPI

The coding for this shouldn't be to to bad...  The UI would be the hard part...

tatarize, 2021-02-04:
This is code-math wise identical to just engraving an image with a gradient. There's no active difference since grey colors do exactly that. 50% grey is 50% ppi, and will engrave exactly like that. I'm pretty sure if you take something partially opaque and you right click the raster operation and select "Make Raster" it will make an image from that, the image consequently will be what would be rastered.

inspectionsbybob, 2021-02-04:
hum... that works for grey (black)

what about other colors like a purple or a green?

thinking about my test card and how to get multiple speeds and ppi with the
least amount of frankenstein-ing.

if i make each speed a primary color, and then fade the color will it
engrave at different PPI levels as it goes along?  Will it factor the
Operations PPI by the fade?

may have to play with that when i get a block of a few hours...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-02-04:
They would be converted towards grayscale based on the average amount of lightness to the color. So purple would be closer to dark gray. Green is a bit more midline.

inspectionsbybob, 2021-02-04:
I was hoping that it would be the opacity of the color that would drive it,
as purple can go from dark royal purple to a light blue sky purple...

need to play with it and see what actually happens... I need time on my
laser which i won't have for a few days since regular work gets in the
way...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-02-08:
Well the opacity would sort of guide it. PPI doesn't differ much from grey color objects. Unless you're turning on the dot length value to something. Which you certainly could do.

closed 2022-07-20 by tatarize