meerk40t/meerk40t #329: Red/Blue/Black raster additional passes not executed

Sophist-UK, 2021-02-05:
I am trying to engrave perspex acrylic and to avoid excess heat melting surrounding plastic when burning an engraving, I want to do 2 passes.

I have added an extra pass, but it is not executed. (Engrave and cut additional passes are executed.)

tatarize, 2021-02-05:
The right click passes thing is a bit screwy and just duplicates whatever is already classified into that routine. So after it's classified. Going to 2 passes will create a duplicate set of objects for that operation. It's basically just making extra sets of objects. And only the current objects within that operation. The passes routine in operation properties->advanced currently is disabled in all versions. So I'm guessing you mean the right click Passes -> 2-Passes.

Sophist-UK, 2021-02-05:
No - I have used Passes -> 2-Passes to duplicate the raster objects - and I can see them duplicated.

But when I execute the job, only the first pass is actually raster engraved - it never returns to the beginning and executes a 2nd pass.

tatarize, 2021-02-05:
Oh, yeah, that's absolutely the case. The raster will not duplicate the paths. They will actually take all of those items you selected turn it into one image and run that 1 time. Yeah, you can make raster than duplicate the raster but that's absolutely the case that the routine I made to do multiple passes there will not work.

Sophist-UK, 2021-02-05:
I can see two resolutions:

1. As you add images to the list to combine, you check if the one you are about to add is a Passes copy of one already added, in which case you skip it and loop around again once you have dealt with the first round of images.
2. Easier solution - disallow Passes->n-Passes for raster objects so users have to use Duplicate instead.

tatarize, 2021-02-05:
Another option would be, if you goto `Passes->n-Passes` it would convert that Raster operation to an Image Operation. Turn that bunch of vectors/images into a single image, and put n-copies in the image in the Image Operation.

Raster could be basically combined with Image operation, if we had a raster group with a step-size within the image operation itself. It would, in theory, make a bit more sense. Since raster groups convert to images and you could duplicate those fairly easily. 

I'm currently working on the tree and how it works in 0.7.0 so I'll look into something like that. But, that would be the more correct way to do it. Have groups and `make image` on the raster, and sort of nest them inside an Image operation. So it's a bit more clear these vector shapes will combine into an image and run like the individual images do in an image operation. Though it might shift things in strange ways.

tatarize, 2021-02-05:
And the shift the operation to an ImageOp and put n-copies in it, would work for 0.6.x where any solution I make for 0.7.0 wouldn't be backported into the stable version since it's too risky that it might create a bug.

closed 2022-02-15 by Sophist-UK