meerk40t/meerk40t #330: Need help building Test document (please)

inspectionsbybob, 2021-02-06:
OK, I give up
I tried to make a SIMPLE test document
3 Boxes
- Speed/Power/Material Test
- Material/Speed 
- Current (MA)

And then 5 lines

100 PPI
_______________________________________  (and a line in yellow)
200 PPI
_______________________________________  (and a line green)
400 PPI
_______________________________________  (and a line in blue)
800 PPI
_______________________________________  (and a line in purple)
1000 PPI
_______________________________________  (and a line in brown)

But when I edit the Color/functions in MeerK40T and then try to run it, I get weirdness.

This is what it looks like in Inkscape.

[image]

All of the  (black) text was going to be 350mm/sec  350PPI

I tried editing it in MeerK40T, saving it, and opening it in Inkscape, a mistake...garbage... just a BLOB.

**Can someone on the team build an SVG file that will do this please, and give me a place to start from/edit please?

Obviously, if you cant edit and save and email an SVG file ready to open to someone, that is a different problem.**

I don't need something FANCY like some of the test prints with lots of text, circles, boxes, and greys...

I just want something that will probably print text and boxes OK (350mm/s, 350PPI) and then draw 5 lines to see what etches and what burns through and can be SAVED, with all the necessary info embedded so it can be opened, scaled to the sample material size, and run without much prep.

Once I get a file that WORKS, I can add boxes and more and attempt to play with Opacity and make it "Fancy."

inspectionsbybob, 2021-02-06:
Here is my starting file...

Please make it work... or get it closer so I can argue with it...

[New Material test.zip](https://github.com/meerk40t/meerk40t/files/5936135/New.Material.test.zip)

tatarize, 2021-02-06:
The main thing is MeerK40t garbles text pretty hard. You need to convert text to paths to do that well.  There's also a lot more line types than that. In theory with 0.7.0 you'd be able to do dashed lines etc.

tatarize, 2021-02-06:
[Bob-Materials-Test.zip](https://github.com/meerk40t/meerk40t/files/5936449/Bob-Materials-Test.zip)

Basically while the lines were colored they also had fills and as long as the raster was there it would add them into raster object. So I turned off the fills in MeerK40t for those objects. Double clicked Hit None, in fills. Then reclassified them all. Got them their own operations.

tatarize, 2021-02-06:
Should probably turn the raster step on the outline thing up a to like 4 so it finishes really fast.

inspectionsbybob, 2021-02-06:
Ok, now you are confusing me...
"Turned off fills in MeerK40t ??" is where, what?
I see a "Hide Fills" under View, but nothing else...
Are you talking about in operations?  There is a lot going on there that isn't documented (yet), I was trying to do most of the work in Inkscape since I couldn't SAVE what I was doing in MeerK40t.

Raster step to 4 makes sense, just low-res the job to get it done fast.

inspectionsbybob, 2021-02-06:
Still working on my first cup of coffee...
Is "Bob-materials-test" an inkscape or Meerk40t file?
(I am still on my tablet trying to do the mornings emails and such before I have to go to real work)

inspectionsbybob, 2021-02-06:
OK, I looked at the file in both InkScape and Meerk40T and am even more confused.

1. The RED Cut Box is gone, no biggie
2. All of the text was converted to PATHs, OK if we are going to ENGRAVE the text...
3. The Fonts had their FILLS set to the color of the line bellow, not sure what that will do
4. The ENGRAVE for the lines is split, there are the engrave jobs for  Yellow, Green, Blue, and then back to BLACK for the X00 PPI text, and then several pages later is the Purple and Brown, why the separation in the operations?
5. I see jobs of BOTH RASTER and ENGRAVE for the same text "100 PPI test" for example. why?  If you converted the text to pure Vectors, shouldn't that be ONLY Engraves?  
6.

tatarize, 2021-02-06:
1. I accidently deleted it.
2. We can still raster the text or engrave it they are paths, not something weird.
3. I dunno what you mean by the fonts those vanished with the conversion to paths.
4. The operations aren't split. They classify in whatever order they fall into the correct buckets. So unless there's some special need they will classify correctly.
5. If there's a raster job at 100 ppi test that's an error. I thought the color was intended to raster at those PPI values and the remaining text was supposed to raster accordingly.

inspectionsbybob, 2021-02-07:
1- ok
2 - Need to play with it...
3 - When I converted a font to Vector/paths I ended up with a lot of
Fragments a pixel or two in size that drove my K40 nuts, going all over the
place to finish a path, and then off to another... Raster would have been
much faster.  Now I picked a weird Curley font, the main path was nice to
watch it engrave, but then it spent minutes dotting "I"s and filling in
dots, and putting the finishing touches on flourishes... never again with
THAT font...
4 - Need to run your file to see what it actually does and then
dissect that...
5 - I will research what is in the operations list...

When you edited this was it in Inkscape, or Meerk40T, or both?  If you
saved from Meerk40T it must have been a later version than 6.17 because 17
won't save...

Raster may be faster for the boxes and plain text, especially if you up the
Step, and then do the lines (or whatever your test symbol ends up being) as
Engrave...

The Rendering of Text is messing me up a bit jumping between Inkscape and
Meerk40T...   K40Whisper uses Inkscape as their Render engine?  I tried to
do something with K40W and it complained it couldn't find Inkscape...
 Need to play more but real work is getting in the way, got called out for
both Saturday -AND- Sunday, by the time I get to the K40 I am tired and a
bit punchy...

Will I be able to save in MK in the next version so I can take MK edits
back into Inkscape?

Bob

tatarize, 2021-02-07:
I left it such that it would raster.
The operations list is the stuff that will actually run. So it's in most ways more important than the elements which is just the shapes and images the data runs on.

Both, in inkscape I converted the text to paths. And in meerk40t I setup the operations. Yes, I used 6.18 because 17 won't save. The version should be done but I'm waiting on Joe to sign off. 

The text rendering in MeerK40t is basically broken, unless you designed the text in MeerK40t it'll be really kinda screwy. Yes, K40W uses Inkscape explicitly to do the rendering, without inkscape it doesn't work. Though it certainly is a kinda reasonable idea of how to fix text is to lean on that code since the font-engine part of it is actually a really hard sort of thing to do correctly.

inspectionsbybob, 2021-02-07:
since inkscape is also open source, can't you find the source of their
rendering engine and pull that in? Or, just call it like k40W does? seems
like a waste of brain cells to try to fix something so it will be as good
as something else that is also open source...

I have only managed to confuse the inkscape engine once, I converted a font
to a path, and ended up with a blob on screen... it printed correctly, but
all the tiny segments gave the renderer fits... K40 didn't like it either
with lots and lots of tiny jerky movements... but it worked...

limited time today thanks to being called out, but I like getting paid, so
today is likely only emails, no K40 machine time...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-02-07:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.18-beta2

Will have the stroke-opacity thing fixed.

tatarize, 2021-02-07:
The engine there is going to be written in C not python, it's pretty easy to reference it. Also, it's licensed GPL and not MIT so that's a non-starter.

inspectionsbybob, 2021-02-08:
there are a few python MIT text engines... OGRE is one i saw come up several times... and a few others.  I can do some research if you give me the constraints such as Python and MIT license...  I know Inskscape also relies on OpenGL, which isn't on older systems, some other rely on DirectX... again, not on older systems.  
The text rendering is an issue that will need addressing at some point, but isn't super high on my list.

tatarize, 2023-08-26:
This was mostly completely subsumed with the Burn Test dialog.

closed 2023-08-26 by tatarize