meerk40t/meerk40t #331: MeerK40T can't open it own save file

inspectionsbybob, 2021-02-06:
I tried to save a workspace (Operations colors and functions)
When I tried to open it again I got:

MeerK40t crash log. Version: 0.6.17
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 1353, in on_click_open
  File "wxMeerK40t.py", line 936, in load
  File "Kernel.py", line 2251, in load
  File "Kernel.py", line 1334, in load
  File "DefaultModules.py", line 152, in load
  File "svgelements.py", line 7458, in parse
  File "svgelements.py", line 7364, in _use_structure_parse
  File "xml\etree\ElementTree.py", line 1228, in iterator
  File "xml\etree\ElementTree.py", line 1275, in _close_and_return_root
  File "<string>", line None
xml.etree.ElementTree.ParseError: unclosed token: line 6, column 648

tatarize, 2021-02-06:
I'd need to see the file, specifically what happens on line 6 around column 648. It's saying the actual XML isn't valid. Like it broke or something or the file is otherwise invalid.

tatarize, 2021-02-06:
The bug found here and in #332 will be corrected in 0.6.18. Which should be ready but joerlane likely dozed off so it'll get pushed a day or so.

closed 2021-02-10 by tatarize