meerk40t/meerk40t #332: Another error msg (may be duplicate)

inspectionsbybob, 2021-02-06:
This was from my attempts to save/open a Meerk40T file and working on the "sample" file

MeerK40t crash log. Version: 0.6.17
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "xml\etree\ElementTree.py", line 1080, in _escape_attrib
TypeError: argument of type 'float' is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "wxMeerK40t.py", line 1360, in on_click_save
  File "wxMeerK40t.py", line 1373, in on_click_save_as
  File "Kernel.py", line 2259, in save
  File "Kernel.py", line 1374, in save
  File "DefaultModules.py", line 125, in save
  File "xml\etree\ElementTree.py", line 777, in write
  File "xml\etree\ElementTree.py", line 942, in _serialize_xml
  File "xml\etree\ElementTree.py", line 934, in _serialize_xml
  File "xml\etree\ElementTree.py", line 1103, in _escape_attrib
  File "xml\etree\ElementTree.py", line 1058, in _raise_serialization_error
TypeError: cannot serialize 0.23137254901960785 (type float)

tatarize, 2021-02-06:
What are you doing? I cannot duplicate these crashes but failure to write that would absolutely also break during the write... That would produce files that could not be loaded since they would not work that way either.

It looks like something that is supposed to be an attribute or a series of attributes but is really just a floating point number and thus cannot serialize. And thus makes a broken file that cannot load. What data is going into this file?

tatarize, 2021-02-06:
Hm. Apparently it should be a string and is a float. I'll double check. Seems legit.

inspectionsbybob, 2021-02-06:
Here is my test file from Inkscape
I loaded it into Meerk40T, and tried to edit something and then do a save and got the error msg shown

[MeerK40t-2021-02-05_21_19_47.txt](https://github.com/meerk40t/meerk40t/files/5936128/MeerK40t-2021-02-05_21_19_47.txt)

[New Material test.zip](https://github.com/meerk40t/meerk40t/files/5936130/New.Material.test.zip)

tatarize, 2021-02-06:
Yeah, that proves the fix I made fixed it. This causes #331 since the save file it will make is broken. Which is why that crash happens.

tatarize, 2021-02-08:
The 0.6.18 version should correct this issue. The corrupted files won't be openable still because they are broken. But, this also prevents #331 accordingly.

closed 2021-02-10 by tatarize