meerk40t/meerk40t #335: Feature Request : Home on START

inspectionsbybob, 2021-02-06:
Depending on what you are doing, having the laser head at the workspace corner (HOME) puts it in the way.

Part #1: I would prefer to have it stay in the corner (Physical 0,0) until the job is about to start, and THEN re-home and start from there.

Part #2: Have the option of where to send the laser head when done, either to MeerK40t home (current option box), or to the corner (physical 0,0) out of the way.

tatarize, 2021-02-06:
Does this idea differ at all from Pre-Home and Post-Home options?

inspectionsbybob, 2021-02-06:
pre/post home options?
The only option I see in 6.17 is to Home after a run on Windows,
Preferences, General Options (Home after job complete)

would likely put the option(s) there...

Are there more options somewhere else??

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-02-06:
Yeah, I removed that option in 0.7.0 since it seems to give that home option more credit than it should. There's a lot of options in start job, in the menubar. That should let you turn some of those things on and off.

inspectionsbybob, 2021-02-07:
I keep missing hidden menus.
Looks like it is there, Home Before....
What is the difference between Home and Origin?   Is there a conflict if
you select BOTH HOME and ORIGIN after?

Wait, wait for what?
Interrupt does what?

Bob

tatarize, 2021-02-07:
I removed the option in preferences since it seemed to be the place to store that. The difference between home and origin is that origin goes back to 0,0 as the program understands 0,0 to be, and home issues the homing routine. There's no issue with doing both. Since it will return to 0,0 and then perform a homing routine. That's not an issue. 

Wait I think issues a 5 second wait. I might be wrong, it's a bit unused and unusable without a set time amount and a countdown thing.

Interrupt stops until you hit a key. It creates a popup you need to interact with in order to spool the next item.

inspectionsbybob, 2021-02-07:
nice...
we need to put those short descriptions in a help or a pop-up...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-02-08:
Closing as the functionality exists already

closed 2021-02-08 by inspectionsbybob