meerk40t/meerk40t #337: Feature request: Help buttons on each window

Sophist-UK, 2021-02-07:
Can we have a help button (question mark in the status bar?) on each window that takes you to the relevant i.e. context-sensitive Wiki page?

tatarize, 2021-02-07:
I'm not sure I follow. There's a question mark in the ribbon bar on the main window. But most of the rest of the windows don't really have status bars to put help icons on.

Sophist-UK, 2021-02-07:
I think I meant title bar. A tiny question mark the size of the minimise/maximise buttons.

tatarize, 2021-02-07:
It looks that way but there's historical reasons they are like that. In 0.6.x the Console.py file refers to the text command inputs and processing. Whereas the Terminal.py refers to the GUI window. If you launch without a GUI, you'd have access to Console but not to the terminal window. In 0.7.0 the console was moved inside the Kernel and the commands were registered in the kernel. This also includes registering help commands for the added commands which can include help commands within the console directly. 

Terminal just refers to the little gui window there to access the Console. The terminology is needlessly confusing but not mismatched. I'll look if ? on the title bars is allowed by wxpython. I wouldn't be very hopeful. OSX might have them but windows doesn't seem to have anything to put in their title bar like that.

tatarize, 2021-02-07:
Doesn't look hopeful, though I could add help, to the ones that already have menu bars pretty easily. Bufferview, CameraInterface, Job Preview. A number of other ones have a right click menu. Namely rasterwizard, keymap, controller (just a weird secret menu), and jobspooler.

Might be plausible to have F1 send any of them to a proper page, but it wouldn't be clear that's what's going on. Also, it would be wx.Python's help so it would theoretically port over to other OSes. I'm not sure if it's not automatically bound or something. There's some OS things where binding the particular help or whatever function does a fairly uniform thing even between OSes.

tatarize, 2021-02-07:
Okay, it's still stuck in the tree recode. Though I could port it out of there if that branch doesn't come back together. But, I can add keybinds to F1 which call a console command of webhelp. Then webhelp provides registered webpages of help. Each window would then be able to launch help with F1. I added it to Preferences to check. There's some minor focus issue, but it also allows webhelp to be accessed otherways. I put it in the gui but in theory it should go somewhere in core. Since it's actually unrelated. 

```
webhelp
----------
Webhelp Registered:
1: help            https://github.com/meerk40t/meerk40t/wiki
2: main            https://github.com/meerk40t/meerk40t
3: issues          https://github.com/meerk40t/meerk40t/issues
----------
```

Then port in the bind for most the windows to call that command. Then you could get window-context webhelp. Though it would be with F1 rather than some more obvious visual cue. And the Wiki pages aren't really that helpful currently, and they'd be semi-hardcoded. Though with the way the plugins work these could be modified or added by plugins.

tatarize, 2021-02-07:
Moved it to core since it didn't actually rely on any gui functionality. And even if you switched the gui or had no gui, you might well still want webhelp. But, it's pretty easy to move that over to 0.7.0 even if the current tree work doesn't fully pan out or needs aborting. Since it's just a little plugin and I registered it in main.py

Sophist-UK, 2021-02-07:
My experience is with Qt / PyQt, and mainly for text-based apps rather than vector/raster graphics apps. I have never used wxPython for GUI - so I have no ideas of its capabilities or issues. But thanks for looking at this so fast,.