meerk40t/meerk40t #338: Graphics (and general) performance

Sophist-UK, 2021-02-07:
I am trying to cut a leather pattern. It is moderately complicated - it has several sub leather pieces each with perhaps a hundred small circles (stitching holes).

On an i7 based laptop (which should have ample CPU power) performance is quite slow - and particularly problematical when trying to do graphic drag and drop operations where you get a lag of literally tens of seconds.

tatarize, 2021-02-07:
A lot of that is the tree being annoying. I'm currently working on that aspect in 0.7.0 and I'm hoping that doing that a bit more correctly and doing lazy icons or letting you not have all of them open, and allowing grouping will make the tree much faster. The scene might still need some other correct optimizations, since it redraws everything every frame and usually shouldn't need to if it stored a bitmap rather than the path object.

Also optimization-wise I might need to perform the animation in an overlay since updating the position of that red circle thing requires everything to be redrawn fairly needlessly.

tatarize, 2022-01-06:
@Sophist-UK 
Is this more or less better? I haven't seen much in the way of speed issues since most of the tree has been finished up?

closed 2022-01-06 by Sophist-UK