meerk40t/meerk40t #339: SVG groups

Sophist-UK, 2021-02-07:
Any chance that the grouping of paths within an SVG can  be replicated in the tree?

tatarize, 2021-02-08:
Yeah, it's part of the reason I'm rewriting the tree. Allowing group and ungroup commands would also provide some proper movement of objects that are grouped etc. So you could have a bunch of paths as a group and use move that around a bit more competently. Also, part of the speed issue for tree is that groups aren't treated correctly since it would be way easier to optimize them at the group level. So if I say drew them onto a mostly transparent image, it get a lot easier to do that if they were grouped and individual parts aren't expected to move without ungrouping.

Sophist-UK, 2021-02-08:
yes - I watched the Lightburn video you linked to in the other issue, and so I understand exactly what you mean.

tatarize, 2021-02-26:
While the rest of the stuff in 0.7.0 is pretty broken, the groups actually does work now. They likely need an icon or something but they exist. And replicate the SVG groups.

Sophist-UK, 2021-02-26:
If I ever get the time I will set up a v0.7 development environment and see if I can help.

tatarize, 2021-02-27:
I'm likely a couple days away from another beta which will have the tree. Maybe sooner if this selections bug is a single easy thing that seems kinda likely. Not that I highly recommend such things. But, it's coming back together pretty nicely.

tatarize, 2021-08-12:
Groups exist as a thing and work rather effectively after the tree upgrade.

closed 2021-08-12 by tatarize