meerk40t/meerk40t #340: SVG path and cutting time optimisation

Sophist-UK, 2021-02-07:
As a longer term goal is there any chance that we can have code to optimise vector path operations. The optimisations I can think of are:

A. When creating the original path list (or when you execute the engrave / cut) deduplicate paths or part paths which are identical - some SVG files have, unbeknown to the user, duplicate paths (which is entirely different to when the user has deliberately created duplicate paths within Meerk40t), and these get unexpectedly doubly engraved / cut. The two use cases I have seen are entire paths which have been duplicated, and e.g. rectangles which are adjacent and share a side (partial or full).

Clearly deduplicating part paths is more complex than deduplicating identical full paths - if this is too complex or time-consuming, then you could perhaps limit this to straight line segments more than a minimum length. Coincident straight lines with different start and end points can be compared for coincidence by checking the following easy algorithm: slope angles the same, at least one end point of one line within the box made by the end points of the other line, that particular end point is on the other line - these are all extremely simple calculations and comparisons. Ask if you think a diagram can help.

B. Merge two paths into one where a first path has the same finish point is the same as either the start or finish point of another path. In other words rather than cut a continuous border which happens to be several segments out of order, try to merge them so that they are cut as one continuous burn.

C1. For **vector engraves**, start with the path with a start or end point nearest the origin (reversing the direction of the burn as necessary). For the next burn select the path with the nearest start or finish point to the end of the first burn. etc. This will minimise the distance of head moves.

C2. For **cuts** use a more complex enhancement of this algorithm which tries to do small cuts first and bigger cuts later i.e. cut the inside cut-outs first and cut the outside of the piece second. However you do not want a literal small-to-large cut sequence because that would take the head all over the place, so you might want an algorithm that e.g. sequences the paths by length, and then collects them into groups of approximately similar size (within a multiple of the smallest path length of the group) and then apply C1 to each of these groups.

I believe that these optimisations will reduce laser elapsed time, and improve vector engraving / cutting quality.

tatarize, 2021-02-08:
Yeah, the current work for the vector path is a bit strange. The instructions exist for the 0.6.x version:

https://www.youtube.com/watch?v=9DQpuke8j_w

There were a couple others in there but you can see the deficits pretty clearly. They require things be merged and they lose some stuff and look weird if they have a fill, #319. This does the main obvious optimization of reducing travel. In some cases the duplicate path would be a desire to go over that area twice. For some cuts it would be proper to remove the duplicated edge. If edges are really close they often get made to overlap with optimize travel and the routine won't stop there. 

B. Gets a little more complex when there would be say 3 points that are coincident there. Since it gets a bit weirder. 

C1. This doesn't actually minimize the distance the head moves. This is actually a pretty naive solution to the travelling salesman problem and generally results in a reduction of something like 30% or so. When a more properly optimized path drops by about 1/10th the previous path. You can fiddle with optimize travel a little bit. I don't have an obvious version of the latter one but you can see it sometimes has longer moves since it optimized all the moves together as a group.

C2. The general rule is you have sub criteria that fit into a larger set of constraints and heuristics. So cutting inner first is constraint since on the cut planning but the heuristic would be to order it such that it reduces head travel. So you make two-opt switches that minimize the heuristic. So if you want to reduce angle changes which is a thing some plotplanning does, and reduce travel you need to specify how much is a directional shift worth vs. an angle shift and then minimize that value. 

Two-opt is an amazing algorithm for that stuff and works well with heuristics and can be made to work with constraints as well.
https://www.youtube.com/watch?v=w1hG1wv8tS4 <-- See lightburn's cut planning.

This set of routines is planned in part with a phase in jobs. In 0.7.0, however it's not done yet, and another feature that is somewhat required for this is also missing, namely preview. It's one thing to optimize things, it's another to show the end-user what was changed and what the current path is. Without doing the latter the former is a bit problematic. Though I'm likely going to do optimize travel as a step even without the preview. Which I think I labeled Project Foresight, but pulled some of those aspects since I wanted a better version of 0.7.0 published sooner. I think I got most of the stuff in there but only published a few buggy betas.

Sophist-UK, 2021-02-08:
C1. I watched the first video and this looks great for vector engraves!! And if there is a GUI-driven or automatic implementation in 7.0 that will be exactly what I am asking for. And yes - travelling salesman is really what is needed.

My only issue with the first video (aside from having to use Terminal which we both understand) is the need to merge and split the paths to make it work.

C2. I noticed from the video that you have an Optimize cut_inner - but haven't found a video demonstrating that from the terminal. But I am guessing that is what I am asking for as well - or rather Optimize cut_travel. But the second Lightburn video illustrates my desires very well. And it sounds like you have it planned - though it is a big job.

(In the end, performance and GUI optimisations are just nice-to-have - Meerk40t seems to have all the basic functionality I need so far to do what I need - yes there is scope for functional improvement, but if I was going to prioritise it would be to improve the documentation - and that is something that I can easily help with.)

If you want help with testing specific parts of 7.0 please ask (though my time availability will be a constraint - who would think that retirement could be so busy even during a lockdown).

Sophist-UK, 2021-02-08:
A couple of things I spotted:

a. I manually sorted the paths and yet a closed curve was not printed continuously because some segments were in the wrong direction. So an easy fix is to check each segment immediately before cutting / vector engraving and reversing the direction if that fixes it.

b. I tried the terminal optimisation and it created duplicate cuts somewhere in the merge / optimise cut_inside / subpath commands. I checked that the original did not have duplicate cuts. I will try to create a small test example with an easily counted set of paths if I get the chance and see if I can narrow this down.

Sophist-UK, 2021-11-14:
Most of this has been implemented so I am closing it.

Path simplification is a whole separate issue to burn optimisation.

closed 2021-11-14 by Sophist-UK