meerk40t/meerk40t #341: Moves BETWEEN vector laser-on paths

Sophist-UK, 2021-02-07:
I noticed today that moves between vector burns were made significantly slower than the burns themselves.

Can we make moves between burns at the same speed as the fastest movement rate of any vector operations?

Explanation: I omit raster operations because I have seen my laser head lose position through excess acceleration or jolt on the Y-axis, even though I can achieve fast speeds on the X-axis for raster burns. And you get vibrations on vector burns which make straight lines wobble that you don't get on raster burns. So I can do raster burns at a far higher rate than I can successfully do vector burns - and that is why head speed for raster burns is omitted when deciding what speed to do laser-off movements between burn paths.

tatarize, 2021-02-07:
You can turn off the Rapid Between and it would move the laser at whatever speed it was currently travelling at, except for rasters. They differ in more than one way, not only are the faster than you'd move a vector move, but they have some extra coding that prevents using them in the same way. They would jump out to rapid mode. 

They wouldn't be at the fastest movement of any vector operation. though I could certainly change speeds and go some other speed but there's some issues with the time it takes to go from one speed to another. It does this twitching thing and that can cost too much speed or not. 

There's also a minimum whereby it'll either run at the current speed to the next object or do a rapid jog to the next location. I think it defaults to 127mils (0.127 in). So if the next location is really close it'll just move there at the current speed.

To properly optimize this it would need to know how quickly the change in speed kicks in, and the speed at which making the travel would be possible. 

---

I could probably add a jog at a specific set speed but, switch in speed is likely going to slow that down enough to make it less optimal. 

---

The suggestion here would likely be part of #258 which notes that rapid can sometimes be bad when doing rotary work. Since you can't actually move the y-axis at the same speed as the x-axis. The moves would need to be rate limited to a specific speed. And likely go at a maximum speed for the specific axis. This speed would basically limit the roll speed between vector paths as well. Though from a different angle. 

Also, I don't know how fast the rapid moves actually travel. But, usually it was faster that most vectoring. But, it doesn't turn off rapid between if rapid between is slower than the current vector moves. Which it quite likely should. I'd guess rapids go about 35mm/s. Though really it's one of those open questions that I've never really bothered to answer.

Sophist-UK, 2021-02-07:
I am not sure that I understood very much of that. Assuming no changes, do I want Rapid Between (which I have had on until now) or not?

tatarize, 2021-02-07:
The moves at a different speed are rapid moves. They involve jumping out to a default mode and moving the laser there. If you want it to not do that Rapid Between off well move the laser within the current mode which will go at the current speed. There's already a cutoff for that with regard to the distance travelled. We do not change modes if the distance travelled is small. The easiest thing would be to have one where if you're in vector mode we don't jump out of mode, if we are fast. It might also be more optimized to not jump out of mode if we are going fast and below a certain distance that varies based on the distance.

127 steps will always just move itself. But, at 20mm/s maybe it should go distances less than an inch. And maybe at 40mm/s it shouldn't jump out to rapid mode at all.

Yeah, I assumed most vector cuts would be at a slower speed hence why I call that other mode rapid mode, but really it could be slower than your actual mode, and if that's the case then it will clearly go slower if you ever jump out of mode. So for some speeds we shouldn't ever use that mode and just stay in vector mode.

Turning Rapid Between will fix your stuff. But, if you have it off and you're doing a slow cut, and it needs to move basically across the entire field, expect to have enough time to write a reply here, before it starts cutting the next section.

Sophist-UK, 2021-02-08:
Hmm - so "Rapid Move" actually means "Rapid Move if you are a snail"? LoL.

Yes - I guess we do need something more intelligent than this.

I was cutting at 15mm/s and the "Rapid" move was slower than this. So, whilst I agree that speed changes are much more complicated, moving at the same speed as burning if it is e.g. more than 10mm/s is an easy win. OR it is easy enough for me to create a test and try to measure the "rapid" move speed from one end of the x-axis to the other so we know what the Rapid Move speed actually is and who when to use it and when not to.

tatarize, 2021-02-09:
Rapid was slower than 15mm/s? What board version are you using? My rapid is a much faster speed than that. More like 30mm/s. 

This makes me think even measuring rapid might not be enough since your note here makes me think they are different speeds.

Sophist-UK, 2021-02-09:
It was my perception, not any form of science - so I may have been wrong. I will try to find the time to take a measurement.

tatarize, 2021-02-10:
@joerlane agrees with you that it's notably slower on his and he too can go faster without the rapids. This really needs some research to see whether switching to a programmed speed and drawing a line or so would save time. However, it is needed  #258 so even if this doesn't end up being faster it might be used anyway to force a threshold on the speed anyway. Because in some cases you need to limit the speed of the y-axis. Which would require speed mode for the jogs *anyway*.

closed 2022-07-20 by tatarize