meerk40t/meerk40t #344: Feature request: outline cut for raster image with offset

Sophist-UK, 2021-02-08:
When engraving an image onto plastic, at the end I want to cut out the image using a positive (for pictures) or negative (for stamps) offset.

Is it possible to make creating a rectangular cut opperation an easy GUI action?

tatarize, 2021-02-09:
Two things you can do:
1) Easier terminal command: `image path`, no care needed.
2) You can drag that image into the cut operation itself. It will actually cut the outline of image.

tatarize, 2021-02-09:
Hm. Actually the offset here makes this a little less easy. I mean I guess you could scale after creating, but yeah that goes back to the whole clever terminal command.

Sophist-UK, 2021-02-09:
A terminal command "image path *offset*" would be better than measuring offset and size, calculating new values with an offset and typing a rect command with these values and hoping you get it right.

tatarize, 2021-02-09:
Seems possible in 0.7.0. Since image path would return an element and then a command offset could maybe give you something there. Though the moving of shapes along the the normal vector isn't done in svgelements even though it would just take some code from svgpathtools to get that done correctly. I guess image path `--offset 2%` or `--offset 1mm` as options would easily doable in 0.7.0 and that's likely how I'd add that. Not sure if that would port over easily to 0.6.x though. But, most feature requests don't get coverage in 0.6.x anyhow.

Also, it could be doable to get alignment tools in the ribbon bar and just draw your rectangle then center the two objects around the same center. Seems like getting those tools online would also correct this, though there again that's not a 0.6.x solution.

tatarize, 2021-02-10:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta2

Added an outline command in 0.6.x. This is bound to control+o this might take `control+alt+shift+escape` to reset all your keybinds. But, `outline` just the bounding box outline. `outline -2mm` outlines minus 2mm. `outline 5% 0%` outlines plus 5% wider on the x and no additional width on the height. Yadda yada.

Sophist-UK, 2021-02-10:
Brilliant!!

closed 2021-02-11 by tatarize