meerk40t/meerk40t #345: Bug: object move function not working

sciroccojim, 2021-02-08:
v0.6.17
MacOS 10.15.7

The "move" arrows don't appear and I'm unable to move the object around on the grid. Arrows don't appear for v0.6.14 either, but I am able to move objects.

joerlane, 2021-02-08:
One thing I noticed today while editing files in MeerK40t, was that occasionally a bounding shape would get selected instead of the object I was trying to select. I would have to de-select that; before I could select the object I desired to move.

It's possible there is an Mac OS version specific issue here; but I haven't been able to reproduce a move bug on my system. I can try harder if there are issues that need to be resolved.

sciroccojim, 2021-02-09:
Thanks so much for looking into this. I just got all set up to record my screen using v0.6.17 and 0.6.14 and “Voilà!” - it works now. Selecton outline is visible (I’m sure it was when it wasn’t working) Yes, no mouse cursor indication but the drag/moving function is now working. Odd. It probably won’t work tomorrow…I’ll update/screen record if it happens again.

Also, is there a way to reliable execute an undo command? CMD +Z only works some of the time, so I have to start over if I select a bounding shape by accident, which does happen.

tatarize, 2021-02-09:
There is no undo. CMD+Z is a lie. It resets the matrix to the original form. Which can sometimes fix things but also can open up cans of worms for no good reason.

Issue #34