meerk40t/meerk40t #346: Ignores "Origin" command at end of Job

inspectionsbybob, 2021-02-08:
Thought I would try out the "start Job" menu items...

Home at start of Job works nicely

"Return to Origin" is ignored, still over the work when finished

joerlane, 2021-02-08:
Tested Return to Origin in 0.6.18 and 0.7.0; and it's working as intended. 

Did some other function conflict with it? I think there may be a bug in recording those settings if the Job window isn't closed first, and re-opened. They are designed to be settings that stay set between jobs, and between sessions (quitting and re-launching MeerK40t).

inspectionsbybob, 2021-02-09:
Been playing a lot today...I went to 
Start Job
Automatic
and clicked on:     
    Home Before     
    Return to Origin After     
    Beep
It shows in the Job list, but doesn't do it...

[image]

joerlane, 2021-02-09:
Where do you expect it to return to?

Home before is the problem? Origin is useful when you unlock the head and move it to a specific location before the job starts. That way; it goes to the same place after. Home before negates this function, and without being able to interrupt AFTER pre-home... That's a process issue.

Also; not 100% sure those settings will stick if you don't close the Job window and Start the job again, and make sure those settings remain. Not sure what version you are running.

inspectionsbybob, 2021-02-09:
I was HOPING that it would go back to the upper left corner, PHYSICAL 0,0

so that it was away from the work area and you could maneuver the blank piece without worry (too much) about hitting the head.

joerlane, 2021-02-09:
Turn off Pre-Home. And set the laser to 0,0 before you click start job. Then return to origin will.

joerlane, 2021-02-09:
I agree. There needs to be support for more Origin points. But: where; and how would they be referenced? I've learned to work within these limits as needed.

joerlane, 2021-02-09:
A HOME, Shifted-HOME, Origin, and physical 0,0 exist. Furthermore. A shifted home can be negative numbers. So a fifth point of reference exists... And more if you jig, or have different beds/index points.

inspectionsbybob, 2021-02-09:
My Suggestion, is to add another option to "Start Job, Automatic" and that would be:

[  ] Return to Corner upon Completion  (Return to PHYSICAL 0,0)

tatarize, 2021-02-10:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta2

To make this less confusing Joe had me add in physical home as an operation there. These should automatically enable and can override your alternative home setting. This also clears another issue that was lacking previous for needing multiple correct alternative home locations.

closed 2021-03-31 by tatarize