meerk40t/meerk40t #347: Mk40 save still corrupting file

inspectionsbybob, 2021-02-08:
Working on trying to build the "Material Sample File" 
 Taking something from Inkscaspe >> MK40 works fine.

Saving really messes the file up....

DEFINATELY can't take it back into Inkscaspe

Looks like you cant edit POSITION in Mk40T, it doesn't place things even close....
This is what I "arranged" in Mk40
[image]
This is what printed out
(The Box fills are likely on my InkScape edits)
[image]

It also injected a few PAUSES/Waits or something where everything "PAUSED" for several seconds... The computer was non-responsive/slow, and then after a few seconds more it picked up and kept going.

It was between different RASTER sections.

Lastly, when it finished the Raser operation, it did the Engrave, but it SLAMMED the cartridge around as if the software expected infinite acceleration/de-acceleration.   SLAM-engrave -SLAM next start point, engrave. SLAM next start, engrave for all 5 engrave lines... 

The only thing "different" on this run is that I pulled the 2nd group of Raser function up higher in the operations list so all of the Raster operations would be done together THEN the Engraving.  Maybe that move caught something it shouldn't have...

Today I have time to "play" with MK40T... 
No error files so far...

inspectionsbybob, 2021-02-08:
I created a new, clean File with ONLY black text and a red cut outline
[image]

When I brought it into Mk40T it looked good
[image]

But when I saved it, not touching anything, and take it BACK to Inkscape to try to edit something
It looked bad already
[image]

I don't care if Mk40 renders text a bit weird, so long as I can take it back to Inkscape to re-edit.

If I have to Keep the Inkscape and Mk40 files separate and parallel it will be a PAIN to edit complex files.

You don't need to even edit anything.  Just open, Save, New, and then re-open the file **I just saved**, and you get the same corruption as when you try to open it in Inkscape, at least it is consistent.

So something other than the Operations stuff and Notes is getting edited/modified or a header is getting corrupted and everything until the NEXT header will be corrupt.

Check the CLOSING tags for Notes and operations stuff where it is getting inserted.

inspectionsbybob, 2021-02-08:
Ok, I created a sample File.
A box with the word "Sample" in it
Saved it in inkscape and PRINTED the SVG file
Opened it in MK40T, Put in "Notes for Sample" in the notes section, Saved THAT, and printed THAT.

Uhm... the two printouts weren't even CLOSE to each other.

PLEASE add some <CRLF>s and formatting.
   Inkscape was easy to follow
TRYING to pull some discrepancies out I saw the following:
Viewspace and dimensions don't agree
Most of the Sections that Inskscspe uses are missing, Defs, Sodipod, metadata, etc... none match up...

It looks like the first portion of the MK40T save file is the settings of the operation menu, but VERY hard to follow without <CRLF>s and SOME formatting
The next section looks like the NOTES
then the two elements should follow, the rectangle and the text "Sample"
Dimensions don't agree for elements
Stroke -Opacity is missing
Transform Matrix values don't agree (Scale?? is this enough precision to scale?)
Text id is missing  (id="text1331")

trying to a "Stare and compare" was useless because of the HUGE structural differences. I tried to find specific elements, and see if they matched values... Most weren't even close.  Since it -is- an SVG file, it should look SIMILAR to the SVG generated by Inkscape... maybe with the Operations and Notes at the front with a header... After the Notes section, everything should look the same!

joerlane, 2021-02-08:
You can bypass the text location issue by editing the file in MeerK40t after saving it in MeerK40t the first time.

So; create a file in Inkscape. Open MeerK40t, select New from File menu (to clear out any existing operations). Open the inkscape file in MeerK40t. Save as in MeerK40t under a different filename. Open the newly saved file. Then move the text objects back where they belong, set any operation powers and speeds, notes, etc.; and save the file in MeerK40t again. 

After you do that; you'll be able to open the file in Inkscape and keep working back and forth between Inkscape and MeerK40t. Even the text will remain text. Changing colors, adding new elements, etc.; should continue to work in MeerK40t following that setup. You may need to manually classify text as a Raster operation, but it will save that property too in MeerK40t.

If you use the attached file; remember the File->New option in MeerK40t. All the extra Operations this file uses will stick around, and may cause confusion or other issues. But generally speaking; this is how I work with MeerK40t on a daily basis. 
[BobTest5_Mk8.svg.zip](https://github.com/meerk40t/meerk40t/files/5947086/BobTest5_Mk8.svg.zip)

inspectionsbybob, 2021-02-08:
I will give that a try... seems odd, but I will give it a try

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

joerlane, 2021-02-08:
The squares on the right are designed to be set at different speeds to see what is required to cut through a material. They hang off the panel edge on purpose, so one can look sideways and see how deep each one penetrated. I often go a bit slower than that on natural wood because its not 100% uniform. This was just cardstock though since it's fast and easy to test with.
 
[image]

inspectionsbybob, 2021-02-09:
OK, tried it, and it is still screwy

Created an Inkscape file
[image]

Brought it into MK40
[image]

saved it under a new name, closed, and opened the new file to:
[image]

Made lots of edits, including getting rid of a fragment, and tried to save, and got an error...

So much for my edits...

Simple stuff works... but complex edits crash, or get completely garbled.

Enclosed is the original Inkscape file, and the Mk40 version
[New GridM.zip](https://github.com/meerk40t/meerk40t/files/5947851/New.GridM.zip)

What I am trying to do, is very close to what you did.  A material file with some standard cuts/text
Dial in the MAX current on the front dial
Pick a Medium high speed (300mm/sec, 300PPI) for Rastering Text 
Set the Step up a notch so it finishes FAST
Print/cut the same thing at a number of different PPI

When I try this,  it garbles things   -AND- it slams the head around as if acceleration is set wrong if I let it run anyway.

joerlane, 2021-02-09:
You've lost parts outside the bed area (why head crashes). Every component needs to be accounted for after reloading the MeerK40t Save as file. 

Usually just work from a template I made months/years ago.... Doing it EVERY single time is not fun. I only draw new when I'm making something new; not trying to laser for the first time. Every project I make requires a few revisions; and MeerK40t settings before it becomes a one click laser job. If you want red,blue,black only jobs; that's pretty much baked in if MeerK40t is setup for that (something you would do on your own).

joerlane, 2021-02-09:
Why PPI? You are going to care about speed between materials? It was even marked on your initial test job (Speed, Material, Current). The file does need to be tuned for each material. Paper does not cut like wood. Acrylic is different still. Each one could be tuned for a 1-click job once you understand those settings for your specific machine.

It can not guess. Running a wood job on paper might catch fire... Sure won't look good.

inspectionsbybob, 2021-02-09:
MOST of the time I set a Medium High Speed, and work DOWN from that.
 If it won't give me the look/cut I want at 1000 PPI I know I need to slow
down by a factor so that PPI will give me the range I want.

The Material Sample sheet was -SUPPOSED- to be a simple way to drop a
sample on the bed and run the job
It was supposed to have enough range to be able to pick the real jobs
parameters from that ONE sample job...

It was also supposed to be a way to learn how MK40 behaved and interacted
with Inkscape and the K40...

In my playing I have worked with Acrylic, cardboard, chip board, thin Birch
plywood, Pine 2x4's. and 1mm hardwood Veneers and even wooden spoons.  I
have been surprised a LOT by what DOESN'T cut easily, and what needs very
very little power to blow through completely.

Not being able to go between MeerK40t and Inkscape is a bother. Not
insurmountable, but a pain.  But when stuff comes across missing text, or
missing attributes, THAT is an error.

On this file, the SAMPLE TEXT all came across as BLACK.  It was supposed to
be the same color as the stars.  Also, the 1000PPI line is MISSING the
"Sample text" line completely... SO... how do I build it in InkScape when
it ignores Inkscape attributes?

tatarize, 2021-02-10:
A considerable amount of this is corrected in the beta. It shows the text in the correct font and doesn't mangle between saves. More work might still be needed to be done. But, this should be significantly in the correct direction.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta2

tatarize, 2021-02-11:
To what degree after 0.6.19 is this still an issue? Leaving this open until that's correctly settled.

Sophist-UK, 2021-02-25:
I saved an SVG on 6.20 and it seemed to reload just fine when previously it didn't. Obviously I was not testing saving of all types of data, so I cannot confirm that it works in all situations.

closed 2022-07-20 by tatarize