meerk40t/meerk40t #349: Convert to Raster Crashes

inspectionsbybob, 2021-02-09:
Clicking on some text and trying to convert to Raster I get....
[image]

The text of the error is:
MeerK40t crash log. Version: 0.6.18
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 2840, in specific
NotImplementedError

tatarize, 2021-02-09:
Hm. That's convert text to vector, it doesn't exist. I'm not sure why that button wasn't removed. The bug is it's not a thing that actually exists.

tatarize, 2021-02-09:
Also, not sure why it says Convert to Raster.  --- I guess it doesn't matter it doesn't actually do anything anyway. But, you can raster text fine, you just can't vectorize it.

closed 2021-02-11 by tatarize