meerk40t/meerk40t #350: Save AS error

inspectionsbybob, 2021-02-09:
Had made major edits and wanted to Save AS a

and got...

MeerK40t crash log. Version: 0.6.18
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 1379, in on_click_save_as
  File "Kernel.py", line 2259, in save
  File "Kernel.py", line 1374, in save
  File "DefaultModules.py", line 80, in save
AttributeError: 'NoneType' object has no attribute 'items'

I had been making lots of moves and edits.
There were some stray "fragments" that got deleted... thinking it was one of those...

tatarize, 2021-02-09:
Hm. The text in question doesn't have a values how odd. But, if you created it with console: `text this is some text` then maybe `scale 50` so you can see it. Then it might not have that. Yeah, seems like a solid bug.

tatarize, 2021-02-09:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta1

closed 2021-02-11 by tatarize