meerk40t/meerk40t #351: Lost Calibration : Need steps to retrain...

inspectionsbybob, 2021-02-09:
Ok, somewhere this last day or so, I lost my scale/size calibration, so everything is off by a percentage, and it gets worse the farther I move away from the origin.

Please tell me what I am doing in the wrong order....  I can't get back to "calibrated" so I am doing something wrong, or in the wrong order.
This is what I am doing
1) Power cycle the K40 so it is "clean". 
2) Put 6x9 printout grid on the workspace
3) Open MK40
    Goto Camera
    Reset Perspective
    Move the points to match the 6x9 corners
    Take several Correction pictures 
    Click "Correct Perspective"
    Goto Windows, Preferences, Bed Dimensions
    Enter 6x9 Grid Dimensions   (mine are 272mm x 142mm )

I have done this several times, and something is off

tatarize, 2021-02-09:
Enter 6x9 grid dimensions? is that supposed to be bed dimensions. I don't get that step. The program doesn't care how large or where that grid is, only that the points between the squares make straight lines so it can figure out the fisheye stuff. Taking the grid pictures corrects "fisheye" not perspective. The corners are the perspective thing.

tatarize, 2021-02-09:
if it gets worse the further you move then it's a linear progressive error meaning your perspective does not match your bed size.

The printout grid needs to be flat, and some pictures are for correcting fisheye. If your camera has fisheye this should be done, otherwise skip that step.

You need to set the bed size next or whatever size you're going to call your bed. Try move 100% 100% to go to the lower right hand corner and "move 0 100%" for the upper left. These spots should be overlapped with your perspective markers. Lining that up based on your bed size should be the best correction there.

inspectionsbybob, 2021-02-09:
Ok, something screwy has happened.
I power cycled the K40, Set HOME as Origin (physical 0,0)
I re-re-re-did the perspective adjustment.
Then I put in my Grid size 246 x 172mm
Then I went to the terminal and typed "Move 0, 100%" burnt a spot
                                                  typed "Move 100%, 100%" and burnt a spot
The distance between them is EXACTLY what I entered in Preferences for Bed Size

When I try to use the CAMERA for placement, however, it is still WAY off.

I just tried an older 6.16 version, to see if it was something introduced in later code, same location

(more testing) 

OK...   I now know it is NOT a scaling issue.
          I took a Simple box.  Pulled it into MK40, made it 100mm X 100mm on the MK40T screen and burned it
          It was exact!  I tried a number of other sizes.  All were right on the money, so I know I got the bed size entered right.

         I then put a big red  "X" on my sample, updated the bed picture, and dragged the box corner to the "X"
         I then used the Navigation "corner" to goto the corner....   It wasn't even close.
         I tried the lower right corner, just in case something got flipped...

[image]

You can see my focusing dot where it thinks the corner is
[image]

I burnt the box, starting from 0,0 physical, and so I could confirm the scale.  Right on the money.
[image]

You can see some other test boxes, they are all the right size, just not where they are supposed to be.

inspectionsbybob, 2021-02-09:
Just tried 6.19... landed in the same place (just wanted to check)

inspectionsbybob, 2021-02-09:
I fixed it, I am not sure how, but it is "Close enough" for work.

I power cycled the K40
I initialized the K40 with K40Whisper
I set the (Windows, Preferences) Bed size to 300 x 200
I put an Oversized piece of paper on the bed
I opened the camera, reset perspective, reset fisheye
opened terminal
      Move 0%, 100% Burnt a dot
     Move 100%,100%, burnt a dot
     Move 0,100%, burnt a dot
     Move to 0,0%, burnt a dot.
Back to camera window
     Set the Perspective markers on the dots.

DID NOT use the grid, DID NOT click Correct perspective

It now is "pretty close" to being "on the dot"  My placement dot and the corner navigation makes it workable.

SOMETHING in the camera routine made it very very hard to re-train the camera scaling., like it didn't want to "take" unless you did things in just the right order.

MANY MANY times I would go through this, using the grid, and it would scale to about 75% of the needed movement FOR PLACEMENT, once there it would get everything right sized, but the placement was way off...

I got it USABLE, so I am not touching that routine again for a while....

inspectionsbybob, 2021-02-15:
Closing this as it is better documented in #359.

closed 2021-02-15 by inspectionsbybob