meerk40t/meerk40t #353: Importing SVG / DXF doesn't take color into account

FabienTregan, 2021-02-09:
I have a dxf with two layers, one red and one blue (#FF000 and #0000FF). When I import or open it (the dxf or an svg export), everything goes in the same job. I can move one element to another job by doubleclicking it, but I can select several element (which would not be practical anyway).
  Finally, even if I split it into two DXF, I can't import both of them in two different jobs, I have to close open the engraving then the cutting, and manage to move everything at the same place (which mean I need to add a bounding rectangle in each dxf).

  Is there some functionnality I overlooked ?

tatarize, 2021-02-10:
Like you want a boundary between the two events without physically making it two different jobs. Like load one, start. File, New. Load the other. Start?

Or am I missing something? Everything that is red gets classified in any red operations, anything blue stroke gets classified into blue operations. If you load multiple files this is still the case. I'm not sure what your trying to do workflow wise?

tatarize, 2021-02-10:
Ah, might mean dxf isn't loading the colors of the layers or something. If that's right I'd need a test dxf that fails.

tatarize, 2021-02-10:
Yeah, saved in Inkscape R14. Did not respect the colors in the dxf.

Will correct.

tatarize, 2021-02-10:
This should be corrected in the published beta. Waiting on full release.

https://github.com/meerk40t/meerk40t/releases/tag/0.6.19-beta2

Also corrected the units, which was kinda a nagging error.

FabienTregan, 2021-02-10:
Thanks ! I tested and it works for DXF. (SVG still load as bold black, but I use DXF anyway :) )

closed 2021-02-11 by tatarize