meerk40t/meerk40t #354: Feature Request : Operations menu

inspectionsbybob, 2021-02-09:
The Operations menu is messy, and long, and only about half of it is really needed.

Suggestions:

1. Add a "Collapse All" option to shrink the list for moving operations around easier
2. Start with files and elements collapsed
3. Keep things collapsed after they are moved
4. Add a "group" option that will AUTOMATICALLY group things by functions, put all of the Raster stuff together, then the engraving, and finally the cut operations.

tatarize, 2021-02-09:
The tree stuff is being worked on currently, since it suffers a series of defects. You're scratching the surface with your list. The work in 0.7.0 is mostly focused on making it dynamic and allowing grouping. So in theory if you load a file it will just modify the tree rather than needing to rebuild the whole thing, and allow it to build the icons dynamically in their own threads because that stuff is slow and causes serious optimization issues.

1. You can collapse that, you just can't tell. Click on it and hit the left arrow on your keyboard. Though any change to the tree will make it expand all again.
2. Files is going to be deprecated. They will be a custom group around elements. So elements -> file -> paths and you could easily access the sub elements.
3. Collapsing isn't something the current tree does well, collapsing things is damned near pointless since they will automatically expand again the next thing you tweak.
4. Grouping stuff is a primary need and part of the reason for the rewrite. There's also functional needs for grouping within the cutplanning. Grouping is also needed to move objects as groups. So if the object is a single thing it shouldn't all need selecting. It should select correctly.
5. Staying collapsed works, starting collapsed might not be useful except for some options like templates (groups of operations that should be switched on fly for different materials etc). It might be worthwhile to save whether the particular operations should be collapsed, but not as part of the startup, but rather as part of the saved file.

inspectionsbybob, 2021-02-09:
All good stuff...
#3, not staying collapsed... a bother, as I was trying to group things so
as to not have a cut in the middle, and it was a pain every time i did
something it insisted on expanding the {unrepeatable} list...  if a user
collapses it, it should STAY collapsed  until they expand it.

my $.02 anyway...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-08-12:
Tree stuff got finished.

closed 2021-08-12 by tatarize