meerk40t/meerk40t #355: Bug: Transparent in PNG is treated as black rather than white

Sophist-UK, 2021-02-10:
If you import a PNG file with a transparent background to raster engrave, the transparent background is treated as black and results in a heavy burn rather than being treated as white with no burn.

I haven't tested what happens when you invert a PNG with a transparent background - but I would personally want it to keep it transparent.

tatarize, 2021-02-11:
As a workaround use raster wizard. It does that stuff correctly. Even if you inverted in the Grayscale step. I'll double check. I did fiddle with some stuff making images but I forced their background to be white or so I thought.

tatarize, 2021-02-11:
I cannot duplicate this.

I bring in a png with transparent background I keep getting consistently white backgrounds. I tried convert to 1 bit. I tried every method of raster wizard. I tried just raw throwing the image unmodified in both an image and raster operation. 

Transparent has a consistent meaning regardless of the the black and white inversion stuff. In rasterwizard you can see how transparent is dealt with. Regardless whether the image is inverted or not, transparent always comes out to be white. Which is always treated as don't fire a laser. When you invert black and white switch because the media gets lighter when laser hits it, but no color absolutely must mean no laser. Elsewise I'd be justified in rastering the entire laser bed except for the white pixels in that raster.

Sophist-UK, 2021-02-11:
I have indeed used raster wizard to convert transparent to white.

When I say it treats it as black, I mean that it burns rather than moves the transparent areas.

tatarize, 2021-02-11:
I managed to get some black pixels after deleting the background where the pixels were not absolutely transparent, but only mostly transparent. Which would give it a hard edge there. And is bugged. I also have reports of an image getting an edge during actualization. https://www.facebook.com/groups/716000085655097/permalink/801756007079504/ I understand what you mean but it seems like it actually would produce black pixels since you can perform all the same operations raster does, except for burning. My test code showed it wasn't burning the entire box that was previously transparent but was move between two sections with transparent between and I guess it could have been burning that, but it seems doubtful. 

Yeah, a video of the steps to reproduce this would be helpful. Maybe the pixels of the original image too. 

I think if rather than 100% transparent they are 99% transparent they would have this issue.

tatarize, 2021-02-11:
Oh. I got the edge there! Yeah, apparently make raster image 2x on an image inside a raster can create an edge. That's confirmed.

Sophist-UK, 2021-02-11:
It does NOT display black pixels on screen for transparent - but it burns deep.

tatarize, 2021-02-11:
Can you make an example. I accept it could be weird form of png or something causing it here and when it gets converted internally is messes things up. I figure if I had an example of it screwing up I could track it down... I'll try Paint.net in a bit likely would make the right kind of file.

tatarize, 2021-02-12:
Paint.net is something else I guess. Not an online paint application. I made changes to fix that Facebook report as well as the little pixels. So there's some chance this will work:

https://github.com/meerk40t/meerk40t/releases/tag/0.6.20-beta1

I'm pretty sure the degree of transparency should directly affect the degree of PPI and thus transparent would be off regardless of other factors but I'll double check the pixel interpretation there.

tatarize, 2021-02-12:
```python
                    def image_filter(pixel):
                        return (1.0 - (pixel[0] + pixel[1] + pixel[2]) / 765.0) * pixel[3] / 255.0
```

That should ... hm. Nah, that's the right order of operations. pixel3 is alpha, and at 255 it'll be 1. At 0 it would be 0. And that would be the resulting power value, 0. So yeah, lower transparencies will result is less image there and permit proper gradients.

So it seems more likely the result of the conversion process which means the Beta1 change is more likely to correct it.

tatarize, 2021-02-25:
The strong belief is that this is fixed in 0.6.20.

closed 2021-02-25 by tatarize

Sophist-UK, 2021-02-25:
Haven't checked, but no reason not to trust that it is fixed.