meerk40t/meerk40t #357: Dxf layer color

FabienTregan, 2021-02-12:
In DXF, it is possible to have a default color for a layer. This is ignored by meerk40t when the dxf is imported.

In the provided test file, the default color of layers "engrave" and "cut" are blue and red, and the line in each layer have default color. They appear blue and red in LibreCad,  but black when imported in meerk40t 0.6.19

```
[layer colors.zip](https://github.com/meerk40t/meerk40t/files/5974523/layer.colors.zip)

tatarize, 2021-02-13:
Okay, read up a bit on the layers uses apparently there's one layer object and the default color and line width can inherit from them. I'll lookup the layer and use that if it exists and isn't overwritten by local a local property which I understand to be the desired procedure for that.

tatarize, 2021-02-13:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.20-beta2

Try that. Should be pretty reasonable. Just has a bylayer check so it shouldn't mess with other file loads.

FabienTregan, 2021-02-13:
> Try that.

I tested, all my files loaded corectly.

Trying to test more picky cases, I found a thing that is not really important but may be corrected:
-Make a cut and an engrave layers with only colors defined (red / blue, width apparently defaults to 0mm at least in freecad but is probably just unset)
-Add a line in each layer (ok: each line appear identical, just the color changes)
-Change the color of the line in cut layer to blue (ok: lines both appear blue. ko: the width of the line in the cut layer uses a different default and appears wider)

I guess this should also apply to line types, but since meerk40t doesn't handle them it has now impact for now :)

Thank you for the great job :) 

[Default value when some other attribut changed.zip](https://github.com/meerk40t/meerk40t/files/5975268/Default.value.when.some.other.attribut.changed.zip)

FabienTregan, 2021-02-13:
I also discover that you can set the color "by group" in addition to "by layer" in dxf :)

tatarize, 2021-02-13:
Hm. I'd likely need to see an example of color by group to implement it.

There are some other attributes, I put in the width earlier and it was kinda annoying and seemed like it risked some stuff messing up, so I set it manually to 1 regardless. The specities of DXF are not my bread and butter.  But, I really don't want to risk adding in line widths with scaling issues since they can easily get screwy. I guess I could flag the layers as non-scaling. But, I'm not really sure the linewidth is entirely 1/10th mm or whatever it seemed to suggest. There's also some criteria like dotted that I could do with an operation but couldn't do as an element itself.

FabienTregan, 2021-02-13:
Indeed tricky. I think I rember 0mm in librecad mean "not set" and will be displayed 1 pixel wide whatever the zoom level is, and this does not change with scaling.

Here is a dxf file with on block, names `my_block`. This block containe two lines with "by layer" color, which appear red in a red layer,  and one line tht has "by clock color", which color will depend on the color set on a particular instance of a block.
The block then appear twice in the drawing : once with no color (then the third line take the layer color and is red), and once with the block color is set to blue (then the two first lines keep the red layer color, and the third one turns blue)
[color by block.zip](https://github.com/meerk40t/meerk40t/files/5975544/color.by.block.zip)

tatarize, 2021-02-13:
ezdxf is spitting out insert entities which are blocks. But, taking a bit more spelunking to figure out how to get the entities I'd draw out of of that. They are apparently like groups. So there should be some way to find these subobjects. I think it's also related to why the R12 output from inkscape didn't give me any responses. They are wrapped in blocks and I'm not looking in those correctly.

tatarize, 2021-02-14:
https://github.com/meerk40t/meerk40t/releases/tag/0.6.20-beta6

Beta6 partially addresses that. Splits the blocks into their virtual entities but somehow still misses the of that last one. LibreCAD says it's Blue. Shows up black, though this is a pretty deep edge case at this point.

closed 2021-08-12 by tatarize