meerk40t/meerk40t #358: Multiple Image Rastering

inspectionsbybob, 2021-02-14:
Got a chance to Play a bit..

Loaded up TWO images that I wanted to burn...

I thought i would raster the entire image at once.  Nope, it Finished the first image (the dragon) and then started on the Kanji.

Took twice as long as I expected as it had to go over the block twice...

No Problem, just a question, is that the optimal way to print multiple images?

[image]

joerlane, 2021-02-14:
If you right click on the Image Operation in the tree you'll get a context menu. From that menu you can select either "Convert Operation->Convert Raster", or Make Raster Image depending on your desired output. 

Both will run as a single raster side-by-side; but "Make Raster Image" will combine the images into one. The "Convert Operation->Convert Raster" will leave two discreet images, but change the operation type to Raster.

inspectionsbybob, 2021-02-15:
More undocumented "hidden" menus with really neat functionality

I saw that more people are submitting documentation requests... Yea.

I really like MeerK40T but the lack of documentation, even basic roll-over
hints is a major hinderance to its adoption by the non-hacker user.

go ahead and close my question, or I will next time I login.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-02-15:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

inspectionsbybob, 2021-02-15:
This is a documentation issue that many have pointed out.

Need to slow down on features and concentrate on Bugs and Documentation.

closed 2021-02-15 by inspectionsbybob