meerk40t/meerk40t #359: Still having Image/camera scaling placement issues

inspectionsbybob, 2021-02-14:
I re-checked my corners and my Preferences 

I am set for 300x200  and my corners at 0,0, 0,100, 100,0, 100,100 are where they are supposed to be.  Dimensions up and down are spot on. 

But when I PLACE SOMETHING, it is WAY off.

 If I burn a 100x100mm box, it is as close to 100x100 as my calipers can measure, but it won't be where it is supposed to be.

I tried placing something as close to the origin as I could get it  (0,0) on screen.

It moved about 1.5" diagonally to place the image.

What can I print/burn to find out where I am off.

The camera, screen and reality don't agree currently, and it is making placing (and trusting the placement) tough.

Here is a video of what I tried
https://drive.google.com/file/d/1HgwgqNx9PtEiW75ExHPAmTMT71Asow70/view?usp=sharing

I created a 40x40 box in Inkscape.  I cleaned it and reduced the drawing to the selecxtion,
making it as clean and small as I could
Pulled it into MK40, aligned it with the box on my target
Told MK40 to go to the corner of the box, which MK40 said it was...
When I look at the laser, it is WAY off the mark.

This is what MK40 is showing me on the screen
[image]

This is where the laser actually -is-
[image]

inspectionsbybob, 2021-02-14:
When I pulled it in the dimensions got garbled a bit, but the CORNER, is what I am looking at, it is still -WAY- off where MK says it is.

joerlane, 2021-02-15:
[image]
[image]
Camera needs to be mounted in a way that is useful, and also proven to be repeatable for the top-down bed view. I need to prop mine open to about 11", and use bubble levels to make sure the lid doesn't flex which would change the point of aim of the camera. These views are both from the same camera; one with the lid closed, and the other with the lid open for a top-down view.
[image]
The 6x9 checkerboard is used to Detect Distortions. This needs to be done for every corner at least once, and the center of the bed. Position, flatness, and lighting are critical for this to work. Do not stop trying if on corner doesn't work. Try a different position, adjust the lighting. Be aware not to move the camera.
[image]
Eight is a decent number of adjustment if they are well spread around the bed. Again; don't cut corners in this process.
[image]
Here is what the image looks like after the adjustment matrix is applied. Notice how all the corners are equally distorted to adjust for the camera lens. If one corner or edge is unlike the rest, it's not good enough.
[image]
The terminal window is opened and the command rect 0 0 100% 100% is typed. After typing the command; press return. This command is used to create a rectangle of the entire size of the bed. This will depend on the bed size being already set accurately in the Preferences beforehand to work. 
[image]
The outline needs to be moved to a cut or engrave operation, and then run on some kind of substrate. I used card stock in this case. It's imperative not to touch or move the material after the outline is marked. 
[image]
Perspective handles are moved to each corner of the marked box as accurately as possible. The dotted lines should explicitly align with the lines made from the outline routine. 
[image]
After checking the Correct Perspective checkbox; it will skew, crop, and adjust aspect ratio (to fit the bed size set in Preferences). This is the usable area of the bed based on the bed size set in Preferences. Both Correct Fisheye, and Correct Perspective need to be checked in order to have a view that is useful to send to the background image of the scene. 
[image]
This is the final background placed. Proven accurate to within a few millimeters. And that too can be tuned by inspecting for minor placement errors, and making fine adjustments to the perspective handles. 

This is written as documentation for the camera setup process, which will be used for the wiki. Please advise if anything is unclear, it can be improved upon, or explained further.

inspectionsbybob, 2021-02-15:
Ok -THAT- got me within mm's.

What WASNT clear before was that you needed so many perspective corrections
-AND- 
You needed to do Fisheye AND Perspective, even if the image looked OK.

My 6x9 grid is almost 180x280, so I didn't think I had to move it around to do the corners.
Again, Documentation and assuming that users know what the programmers had in mind. 

WE CAN'T READ MINDS !!

Additional Notes to add to Set-up/Calibration Wiki

**Materials NEEDED**
10"x13" piece of heavy card stock paper or Carboard.  
    (It should be LARGER than 200mmx300mm by at least 25mm on all sides)
Sharpie pen
6x9 Grid printed (size recommendation)

**Good description of how to burn corner marks**
"Circle 100% 100% 1mm" isn't the complete command sequence. It just creates an operation object; it doesn't burn the circle or even move to that location... some other commands are needed.  Similarly, RECT 100% 100% just creates an Operations object. It doesn't burn the box.   Will need to tell people to adjust the power for the material they plan to Mark; at 100% power, it will make a mess of paper or cardboard, you need to either do a "Power 10%" (or whatever the command is) or tell people to turn down their front panel power. 

**Steps   (with your pictures)**
- Reset K40
- Set HOME to 0,0
- Reset Perspective & Fisheye and make sure boxes are clear
- Use the 6x9 Grid in the Corners (2x GOOD images in each corner is recommended)
- Turn on  "Correct Fisheye"
- Turn on "Correct perspective."
- Mark the corners by doing
-   Move 0% 0% 
-   Test Button to mark
-   Move 0% 100% 
-   Test Button to mark                                         (Or provide a **COMPLETE** script of how to burn the corners)
-  Move 100% 0% 
-  Test Button  to mark
-  Move 100% 100% 
-   Test Button to mark
- HIGHLIGHT THE CORNERS WITH SHARPIE if you can't see the marks clearly on the Camera
- Move the Perspective Corners to the Marked Corners
 
Note: You can "Full Screen" the Window and zoom to help get the placement exact.

This SEEMS to work.   Get a lot of "ill-conditioned Matrix, Keep trying" though, not sure what that actually means.

joerlane, 2021-02-15:
The wiki has already been updated with a number of possible way to do the setup (and probably more to be added). Not everyone will have a specific set of materials on hand, or even the same bed setup as a starting point. No assumptions can be made; and many specifics only apply to individual use cases, not a generic set of setup and use guidelines. 

The instructions provided say: "_The outline needs to be moved to a cut or engrave operation_". There is a way to do that part from the terminal (more automatic), but that burden is placed on the user explicitly; and for good reason... An Operation choice, and settings for marking that rect would depend on the material used as you noted. Even valid Red or Blue operations, or user specific settings can not be assumed. Lack of automatic classification furthermore places the burden of running the laser on the user, instead of trying to fire the laser automatically without the user understanding what is going on. Even simply placing the rect into an existing operation might go unnoticed by the user; that could cause a head crash, or laser firing on the smoke vent if other setup constraints are also ignored. This was very much on purpose (despite having tested many other options which "seem" easier).

A bunch of things are problematic with that bulleted list. Instructions can't require 0,0 home, not every machine can use that setup; most notably the stock smoke vent is in the laser path. Resetting Fisheye might not be an appropriate instruction; since some users might not need or wish to delete the matrix they already have (it would already be saved if it had been created by the user in the past). It is fine to overwrite that matrix with a new setup. The script for the complete burn "corners" was provided (rect 0 0 100% 100%). There are in fact more complex aliases that draw alignment marks only in each corner, and not the full rectangle specified in the instructions. But those aliases are even less easy to understand in terms of intent, since they contain a dozen or more commands. It's been tested; and an alias command can even do the HOME, as well as place the alignment marks in an Operation ready to run. That is simply not done for the reasons stated above (not safe, and the user needs to be involved as well as follow every one of the instructions). The sharpie wouldn't be needed if you used the rect 0 0 100% 100% command. The rect 0 0 100% 100% was chosen since the lines are visible with even low-res or blurry cameras (burning tiny dots with the laser test are not likely to be visible in the camera view, even when zoomed in).

Indeed; you can zoom in to get really in depth with the perspective corner markings, but for the screen shots that would have been difficult to show clearly. I also did it zoomed out to see how terrible it could be, and it worked just fine. In my experience, it's nice to do some fine tuning after further testing anyway. Doing a manual adjustment of those handles later to correct minor placement issues can allow for sub-millimeter accuracy. Quite frankly though; the stability of the camera mounting location (lid flex, repeatable opening position, etc.) is likely to have more effect on alignment than fine tuning the perspective handles.

The Distortion Detection routine is rather complicated. It's analyzing the camera image to find the 6x9 grid, and use it to warp the image in a favorable way. If that detection, or the math goes wrong; it won't intentionally record garbage data. Yes, it's difficult. You may need to nudge the checkerboard around or adjust the lighting. But there really isn't a way to do that setup manually. I found that rotating the checkerboard helped sometimes, but other times I just moved on, and came back to address a particular location later with success.

joerlane, 2021-02-15:
One other reason I avoided recommending the Test Fire at four corners, and a Sharpie marker to highlight them; is it adds an opportunity to bump things out of alignment before the perspective is set. It's an avoidable point of failure, and one that could cause frustration for an end user. That was also mentioned in the instructions explicitly: "_It's imperative not to touch or move the material after the outline is marked._".

inspectionsbybob, 2021-02-15:
Ok, all makes sense, especially the vent being in the way of the laser on
the traverse that crosses it.

HOWEVER:   RECT 0 0 100% 100% doesn't burn the rectangle by its-self.
Neither does Circle 100% 100% 1mm
 If I could have used those commands, I would have as I could do that with
the cover down and not touch anything for 8-10 commands.  There is some
OTHER command that is needed to execute the burn of the marks.

ASSUMING that the user -HAS- put down something appropriate to serve as a
target
what would be the sequence
Circle 0 100% 1mm
Circle 100% 100% 1mm
Circle 100% 0 1mm
Circle 0 0 1mm
Power 30%      < or something so that it just marks paper/cardboard and
doesn't blow through it and torch things
<what goes here to tell it to burn the circles the above instructions put
onto the operations list?>

If you aren't comfortable writing these, then -ALL- reference to this needs
to be removed so the non-hacker user doesn't start a fire or hurt
something.  The "Move x% y% and then test button"  then needs to be well
documented, very much like you did for me...  (probably safer. maybe an
"advanced" section with the command/script in THAT section)

I apologize for dragging you through this (again), but I am not a hacker
any more.  The K40 stuff is my 4th or 5th priority, and not anything
related to my day job, other than I would like to make some giveaways
someday and I enjoy tinkering.  My "Z" axis bed is just waiting on a GT2
belt and it will get installed... Probably will never use it much, but it
was fun to build and debug...

Bob Sisson

inspectionsbybob, 2021-02-15:
As for things not moving, I use strong MAGNETS to hold stuff down.  It will
hold to the frame, or if you have installed a honeycomb bed, they will
stick to that as well.

Bob

joerlane, 2021-02-15:
The circle command doesn't give an appropriate guide for the perspective handles; in fact, it is likely to be used very incorrectly, generating significant placement error. Those circles wouldn't be drawn outside of the cutting area, and thus the perspective handles would need to mark a point with no reference at all based on circular markings. The corners need to be marked with at least a line, or a rectangle. The fully bounded rectangle has significant advantages in this situation.

Yes; the object is still not put into an operation; on purpose. Doing so makes the terminal commands more complicated, but also makes the instructions considerably more confusing. The instructions simply can't explain every possible MeerK40t setup. It can't even assume basic things, like Red is always Cut; since the user can configure all those themselves (or not have done any configuration at all). Can't assume Blue/Engrave either; maybe the user has only don't cuttings jobs with Red/Cut. Instructing the user to move the recently created rect 0 0 100% 100% into a cut or engrave operation; _is the instruction for that process_; that absolutely requires the end user to make some specific and necessary decisions. The newly created rect needs to be dragged from elements, where it appeared; into an appropriate operation... The user performing that action should be using their own knowledge of their setup to make that happen.

You mention power needs to be dictated in the instructions again; which would be a HUGE problem for individual users. That "30%" setting on a k40 with digital controls could amount to more than 100% of the safe laser power on some machines. Similarly; lasers with analog power control also don't have a 30% setting, or any safe percentage, or knob position either. That's entirely machine specific (and something that every user should have learned about before turning the laser on). Even a current draw value, is not a sound metric to be included in the instructions. 

There is no range of default settings that can be shared universally. Things like air assist (which aren't a default or stock install on a k40) dramatically change even the most basic guidelines for power and speed on a given material; not to even consider variables like orifice size of that air assist nozzle, distance from the work, the aim point of the nozzle, or the pressure.

It's obviously a work in progress. But I've spent considerable effort to be agnostic about documentation for the camera setup. Many things were learned in the process; even noting that "stock" dimensions of any specific k40 machine are not a given. Some machines even have a physically smaller bed than others. In other cases; a user has already made accommodations for the bed position (shift), and size. Those things need to be universally accepted and respected. Giving too many instructions can be just as dangerous as too few.

Rest assured, every suggestion, issue, and potential use case is considered carefully. Fully automatic operations have already been tested. Ultimately it came back to the same thing described earlier; that  instructions need to work for every user, and every use case. The goal of documentation for any process should also be; "First, do no harm". Especially so with lasers involved.

inspectionsbybob, 2021-02-17:
FIXED !

closed 2021-02-17 by inspectionsbybob