meerk40t/meerk40t #361: Enhancement: Open Windows where (and size)  they were closed

inspectionsbybob, 2021-02-16:
Simple, maybe...
Landscape is limited on laptops, and once someone has gotten their windows they way they want them so everything they need is showing and is sized right, once they close a window (camera for example) why not re-open it in the place it was and the size it was so you don't have to drag it back to where it was, and possibly re-size it as well.

tatarize, 2021-02-16:
I recently got it centering the main window again, there's some worry this will have issues if those settings get corrupted putting windows in absurdist and unreachable places. It might be unfounded, but it's still a worry.

inspectionsbybob, 2021-02-16:
The solution is to put some variables in a file that can be deleted if it
become corrupt.  If the file exists, use the values in the file, if the
file doesn't exist, use the default values and rebuild the file.

I understand the attraction of a program that can be run without any
drivers or loading anything, but at some point people are going to want to
save more that can be saved they way you are now between sessions.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-05-20:
This is now part of 0.7.0.

closed 2021-05-20 by tatarize