meerk40t/meerk40t #363: Crash report

inspectionsbybob, 2021-02-16:
MeerK40t crash log. Version: 0.6.16
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "Kernel.py", line 1725, in run
  File "Kernel.py", line 1923, in run
  File "Kernel.py", line 1790, in shutdown
  File "Kernel.py", line 2182, in close
  File "Kernel.py", line 100, in detach
  File "Navigation.py", line 371, in finalize
RuntimeError: wrapped C/C++ object of type SpinCtrlDouble has been deleted

tatarize, 2021-02-17:
Um. That's version 6.16. Seems like you might want to upgrade. I think that bit of code got changed accordingly.

We're at 0.6.19 for regular releases and 0.6.20 pretty deep in the betas.

inspectionsbybob, 2021-02-17:
Closed fixed

closed 2021-02-17 by inspectionsbybob