meerk40t/meerk40t #371: want to be able to HIDE the Spooler (or not open it)

inspectionsbybob, 2021-02-19:
The spooler is nice, I think.
But when it pops up when a job starts, it hides the pause button, and the controller tab (where the only STOP button is).

Can we have an option to HIDE THE SPOOLER? 

landscape is short enough without the spooler taking a good amount of it for nothing...

There is nothing on the spooler page that isn't on the Job start page, and the Job start page disappears.  I use the Job Start page to confirm that i have everything set right, no stray Raster, engrave or cut operations.  I can always click on the spooler to open it, but that is really rare.  Once a job is running, it either finishes, or is aborted via the Controller page.

joerlane, 2021-02-19:
There is an option for this in Settings. Uncheck "Launch Spooler JobStart".

inspectionsbybob, 2021-02-19:
have to tripple check, but i thought it was already undone..l

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-02-19:
closed as the feature is there

closed 2021-02-19 by inspectionsbybob