meerk40t/meerk40t #373: question: is there a "running" lead anywhere?

inspectionsbybob, 2021-02-20:
Is there any place on the controller that signals that it is "running" a job?

I am trying to find a place to tie into for a number of external components such as "air assist" that should be turned on when the K40 has the laser turned on.

I would LIKE to turn the air assist off when not running to keep it slightly quieter in the shop area, and turn it back on when a job is running.

Not really a Meerk40t question, but I figured someone on the group would know...

joerlane, 2021-02-20:
The controller board hard a connector with pins labeled EX+, and EX-. That is able to control a solid state relay directly; which can be used to switch peripherals on and off with the laser job. It can turn things on when the job starts, and will run an additional 8 seconds or so after the job finishes. 

I got a few Inkbird SSR-40 DA (DA stands for D/C - A/C, as they are available in different configurations for those control and switching requirements). I bought a few of them for $5 each a while back with free shipping. You might pay a bit more if you want them fast (I think these took about 2 weeks to arrive from China).

joerlane, 2021-02-20:
In case it wasn't clear enough. The SSR-(##) DA version is the correct one. The D/C side is for the control signaled by the EX+/EX-; signal switches the solid state relay contacts capable of dealing with an A/C load. 

It's somewhat possible you might want to switch a D/C load; and you could use a SSR-(Current rating) DD instead. But either of the AA and AD models would not work since the EX+/- pins are D/C.

Of course you could use other brands, with different model numbers as well; as long as they meet those desired specifications as needed.

inspectionsbybob, 2021-02-20:
fantastic!

Please don't wait for me to respond when booking an inspection. I turn off
my devices when on appointments and someone else could get your spot.
Please CALL our service or book yourself online to save the spot as soon as
you know the time you want.

Bob Sisson ACI, BVI
Inspections by Bob
ASHI Member # 212016
MD Home inspectors License #29666*
*MAC-ASHI Chapter President 2010-2012
ASHI Mid-Atlantic Group Leader 2009-2012
ASHI National Board Of Directors 2012-2015, 2017,2018-2020
Bob@Inspectionsbybob.com
www.inspectionsbybob.com
(301) 208-8289 Scheduling/Pricing

inspectionsbybob, 2021-02-25:
Sorry, one last question about this:

Are you going directly from the Nano-controller Ex+/Ex- to the SSR?

joerlane, 2021-02-26:
Yes; directly connected.

The M2:Nano controller has a bit of isolation on the EX+/- port (a transistor protecting the controller chip); and the SSR is fully opto-isolated internally as well. 

That is the big reason for the labeling of the polarity on both EX+/- pins, and the SSR as well. The opto-trigger (internal to the SSR) is an LED, so that is polarity sensitive. You'll need to connect the + connection on the EX port with the + pin on the SSR input trigger, and match both negatives as well. 

I had some temperature probes that happened to have the mating connector for the EX+/- jack. Simply chopped it off, and screwed it in to the SSR. It's a standard pin size/spacing. Shouldn't be difficult to source one if it can't be easily harvested from something you have laying around. Attached the datasheet for the SSR as well.

[image]
[SSR-10DA-Fotek.pdf](https://github.com/meerk40t/meerk40t/files/6047243/SSR-10DA-Fotek.pdf)

inspectionsbybob, 2021-03-11:
Got it working ! My issue now is the leakage current is more than enough to keep the tiny air pump I use running.  Will have to find an SSR with less leakage current.

Thanks for the info.  I will close this one,

closed 2021-03-11 by inspectionsbybob