meerk40t/meerk40t #374: 6.19 does -NOT- work with default windows driver

inspectionsbybob, 2021-02-21:
I just had a computer crash - update failed and took the boot sector, cost me half a day.  Anyway.

Had to rebuild from blank.
Thought I would try it with the default (Windows) driver.
Nope... Nothing. Nada.
Loaded the K40Whisper Driver
Everything works great. Found the K40 instantly.

PS... Re-Calibrating and re-perspective-ing I have down to less than 5min.  Ran a test burn, Spot on!
Lost the camera several times, had several "pauses" but the job ran perfectly otherwise....

Now I get to re-install all of the OTHER stuff...

When will 6.20 be available?

tatarize, 2021-02-21:
I pretty commonly run it with the default driver. I test it rather commonly. It needs to detect and have windows update or whatever find the CH341 driver and install it. If nothing installed then it doesn't really work. But, it tends to find the driver and hooks it up then it works fairly fine.

---

How do you keep losing the camera? I mean the code doesn't change. Does the camera move at times or something? 

---

0.6.20 we're muddling through some release candidate stuff. There's the betas in the releases but we're dealing with some minor issues that you can see on the PR
https://github.com/meerk40t/meerk40t/pull/356 -- Usually we don't stray too far with code changes because the goal is to abandon the 0.6.x line for 0.7.x line. Any features have to be ported into both, but fixes are always fine. This time a bunch of things might have changed as you can tell from the last beta's change log. https://github.com/meerk40t/meerk40t/releases/tag/0.6.20-beta7

All that needs to be tested to avoid introducing new bugs.  Though it *should* be fairly fine. Though adding in Centerize DXF #273 is a took a little bit. And dynamic checkboxes for the pan and zoom changes for mac should be fine, but 0.6.x is the stable line so changes should be kept minimal.

So it could be a day or a week. Depending on whether things pass the various checks.

inspectionsbybob, 2021-02-21:
i have the camera running during the job, and it often will say, in the
middle of the job, "no webcam found". At that point I need to close the
camera window and open it again to get the camera back. The job is still
running...

What i have noticed is that the "pauses" often coincide with loosing the
camera...
If I turn off the camera I also see less pauses.  No errors, and no log
(that I know about) so I don't have anything concrete to tell you.

As for driver... when i rebuilt windows and powered up, it said it found
the unknown device and installed something... but it didn't work.  I was
short on time so I loaded K40W's driver and never looked back.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-02-25:
Just loaded 6.20 up 
Haven't seen a "No Web-Camera Found" message yet...
Seems to have fixed it.  I will close it soon if the USB links seems happier

closed 2021-03-31 by tatarize