meerk40t/meerk40t #378: feature request: Acceleration control

inspectionsbybob, 2021-02-24:
i was trying to cut out some thin material and I needed to run FAST because the material was so thin.  Even backing down the power I had problems with burning. If I ran faster, I could use more current, which seemed more stable, but the K40 didn't like cutting at those speeds. sounded like I was beating on the poor thing. 

Is the acceleration/de-acceleration  settings different for Raster than cutting? The back and forth of the raster is smooth, but fast movements while cutting is jerky and clunky.

Yes, I could, and did, end up using PPI to lower the effective power, but I still wonder about the thrashing of the K40 when cutting at speed vs rastering.  Same thing for engraving, seems to thrush/clunk much more...

inspectionsbybob, 2021-02-24:
I do have a vinyl cutter that can do really thin stuff, but I really hate weeding.  The laser is a slower to cut, but no weeding, and the camera lets me position the job to use my cutoffs rather than trashing them.

tatarize, 2021-02-25:
The only seeming acceleration stuff is already given access. If you double click on an operation, check advanced. There's an acceleration value. This sets a value between 1 and 4. This is in turn sent to the board in the speed code for that operation. Now, there is *some* chance that this might actually let you fine tune those speeds to work better but there's also some chance your machine just can't do those speeds.

Codewise however, there's nothing else I can do. I give to the end-user everything that can be set on the M2 Nano.

tatarize, 2021-02-25:
Also, though you said you had a vinyl cutter and you're cutting "thin stuff". Do not cut vinyl. (Not sure you are but anytime those are mentioned together the running point is to tell people to avoid creating death gas caused by hitting vinyl with lasers).

inspectionsbybob, 2021-02-25:
Don't ever cut vinyl, I know better.  I have done some neat hybrid projects using both though...

Is 4 the highest acceleration?  What is the default?

tatarize, 2021-02-26:
The default depends on the speed.

Non-Raster:
1: [0 - 25.4]
2: (25.4 - 60]
3: (60 - 127)
4: [127 - max)

Raster:
1: [0 - 25.4]
2: (25.4 - 127]
3: [127 - 320)
4: [320 - max)

tatarize, 2021-02-26:
These are taken from the chinese software itself. There's an open question as to whether you can get better results going higher in the accel factor sooner. Like you can go at 25 and it's the highest edge of useable. But, you can also go at 60 and your machine will basically seem like it's tearing itself apart. So if you went at 50 but in 3 or 4 might that work since that is the high vector work anyway.

But, solving that is hard and would take a lot of trial and error.

closed 2021-08-12 by tatarize