meerk40t/meerk40t #382: 0.7.0 beta 11 selected path in operations t highlighted in scrf

kgwatson, 2021-03-08:


tatarize, 2021-03-08:
Confirmed. I vaguely remember Joe telling me that but I was getting a lot of bugs fixed at the time and neglected correcting that. It shouldn't be too hard to correct.

tatarize, 2021-03-08:
Well, I did it for the selected path highlighting in scene. But, not right off for selecting the operation should select all the children within that operation that I assume is also desired. https://github.com/meerk40t/meerk40t/commit/bcd73d7a098503d2feb7b79a42ec9553e7ce4612

tatarize, 2021-03-08:
I'll publish beta 12 with the rotary fixes and that selection fix after I get through these crash logs.

tatarize, 2021-03-08:
https://github.com/meerk40t/meerk40t/releases/tag/0.7.0-beta12

closed 2021-03-31 by tatarize