meerk40t/meerk40t #383: Duplicate Raser Operation IGNORED   (v 6.20)

inspectionsbybob, 2021-03-09:
I was trying out different materials (again) and wanted to see what a second pass would do...
So I "duplicated" the run...

It was IGNORED.   I tried several times, it only made ONE pass each time...

[image]

tatarize, 2021-03-10:
Yeah, it's not ignored. it's actually just combined into a single image. You'd have to use Image as the operation to do that. Since every copy of that in there just gets duplicated and rendered into the same image. See #329. I am still not sure how to do this correctly. At least in a way that makes it clear what's going on here. Though in 0.7.0 multiple passes of the Image operation would execute.

joerlane, 2021-03-10:
This works if you create another Image operation, and then drag the duplicate image in that new operation. 

If you right click Operations, "Add Operation" will create an "Unknown" operation at the bottom of the list. Double click on that, and select Image as the operation type. Then drag the duplicated image into that. A new operation also needs the speed, PPI, overscan, etc. set. That is in fact a feature; as it also allows you more flexibility on the second pass. Of course you can Enable/Disable each of those Image operations as you see fit to customize a multi-pass job.
[image]

inspectionsbybob, 2021-03-10:
but what I was trying to do, was take an existing Raster operation, that
had a speed and power already assigned to it, and just do it again, what
most would think Duplicate would be. It may be a feature, but it is
misleading. Duplicate, add passes, I can see mostly for CUTTING, I will
need to verify that works the way i expect it to.  6mm Poplar plywood takes
3-passes to cut unless I want to up the front panel power.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2021-03-10:
> In 0.7.0 there's a passes option you can modify in operations which would
> do this correctly. Nobody's saying you're wrong. You're totally right, this
> is clearly defective behavior. I just don't have a good method of
> correcting it. I guess the best thing to do would be to remove the option.
> If you are dealing with a raster, you should not get the option of passes
> or duplicate since it's grossly misleading. That'll probably be the
> solution in 0.6.x. The same would be true for 0.7.x but there I could
> duplicate the node pretty easily and show what's going on better, but the
> way forward is basically removing those options in 0.6.x.

i am working ANOTHER beta as well for my 3D printer, 90% of issues are
related to DOCUMENTATION and telling people how things work, or should work
and managing expectations...

sometimes it is easier to put a note in the menu, or a hover-note, than to
fix the code in the current release path.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
bob@inspectionsbybob.com
MD Lic#29666; ASHI Member #212016

tatarize, 2021-03-10:
You will no longer be able to run Passes on a Raster Operation in 0.6.22. Since that function created additional object to be rendered to an image, it did not create passes.

tatarize, 2021-03-10:
All the stuff inside raster is turned into 1 image, then rastered. Making additional copies does not make it turn into more than 1 image. So it does not actually do passes, it simply adds more stuff to be turned into 1 image. That option was confusing and pointless and is thus removed.

However your original picture has duplicated images and not rasters, that will actually work fine. If you ran the stuff in your image, it would work. However I assume you previously did that in a "Raster" operation, and there it would fail completely. I assume the image is created as a mockup of the other problem you experienced with Raster which absolutely has that defect exactly as you described.

closed 2021-03-31 by tatarize