meerk40t/meerk40t #385: Feature Request : Add basic camera controls

inspectionsbybob, 2021-03-10:
I would like to see BASIC camera controls for Brightness & Contrast, and MAYBE Focus.

The DEFAULT settings on my Camera are too Bright, so the image is washed out.  Also the Autofocus on the camera gets confused with the Laser Head, rather than the material.  Focus may not be available on all cameras.

tatarize, 2021-03-10:
```python
cap = cv2.VideoCapture(1) # my webcam
cap.set(3, 1280) # set the resolution
cap.set(4, 720)
cap.set(cv2.CAP_PROP_AUTOFOCUS, 0) # turn the autofocus off
```

This is apparently a thing. I am reluctant to make the camera GUI overly feature bound since you can usually modify those settings in the driver but I could adding some of these things in 0.7.0 if they have fairly reasonable methods to access those things. 

```
CV_CAP_PROP_POS_MSEC Current position of the video file in milliseconds.
CV_CAP_PROP_POS_FRAMES 0-based index of the frame to be decoded/captured next.
CV_CAP_PROP_POS_AVI_RATIO Relative position of the video file: 0 - start of the film, 1 - end of the film.
CV_CAP_PROP_FRAME_WIDTH Width of the frames in the video stream.
CV_CAP_PROP_FRAME_HEIGHT Height of the frames in the video stream.
CV_CAP_PROP_FPS Frame rate.
CV_CAP_PROP_FOURCC 4-character code of codec.
CV_CAP_PROP_FRAME_COUNT Number of frames in the video file.
CV_CAP_PROP_FORMAT Format of the Mat objects returned by retrieve() .
CV_CAP_PROP_MODE Backend-specific value indicating the current capture mode.
CV_CAP_PROP_BRIGHTNESS Brightness of the image (only for cameras).
CV_CAP_PROP_CONTRAST Contrast of the image (only for cameras).
CV_CAP_PROP_SATURATION Saturation of the image (only for cameras).
CV_CAP_PROP_HUE Hue of the image (only for cameras).
CV_CAP_PROP_GAIN Gain of the image (only for cameras).
CV_CAP_PROP_EXPOSURE Exposure (only for cameras).
CV_CAP_PROP_CONVERT_RGB Boolean flags indicating whether images should be converted to RGB.
CV_CAP_PROP_WHITE_BALANCE_U The U value of the whitebalance setting (note: only supported by DC1394 v 2.x backend currently)
CV_CAP_PROP_WHITE_BALANCE_V The V value of the whitebalance setting (note: only supported by DC1394 v 2.x backend currently)
CV_CAP_PROP_RECTIFICATION Rectification flag for stereo cameras (note: only supported by DC1394 v 2.x backend currently)
CV_CAP_PROP_ISO_SPEED The ISO speed of the camera (note: only supported by DC1394 v 2.x backend currently)
CV_CAP_PROP_BUFFERSIZE Amount of frames stored in internal buffer memory (note: only supported by DC1394 v 2.x backend currently)
value – Value of the property.
```

Some of these might work quite well and it wouldn't be too outlandish to give them access through the console. So like `camera0 set fps <value>` though this might take some actual testing on some cameras to see what works or rather whether setting those values might crash things or cause instability.  Then if that works well maybe consider giving some of the more useful well regarded options some gui support.

This actually seems like a pretty solid idea with some strong backing.

inspectionsbybob, 2021-03-11:
I would probably go for:
   CV_CAP_PROP_BRIGHTNESS
   CV_CAP_PROP_CONTRAST
and I think you are already using
  CV_CAP_PROP_FPS Frame rate

Bob

tatarize, 2021-03-11:
I actually make my own framerate there. In 0.6.x it's by asking for the images at certain intervals and in 0.7.x it's by having the background process hammer for frames and the foreground process retrieve the latest frame at whatever interval it can. It might be kinda lossy on that. Really a lot of stuff there would be fine to add. It would be a line or two on my end in the console to trigger those. Even if I'm not really sure the operands they take it it's not too hard in 0.7.0 to do that stuff.

inspectionsbybob, 2021-05-13:
Would -like- sliders for
Brightness
Contrast
Focus maybe with a [X] for turning OFF autofocus as it gets confused easily...
you already have frame rate as a slider.

inspectionsbybob, 2021-05-13:
Ooops....
This page Might Help, looks like OpenCV has some updates including focus control
https://github.com/opencv/opencv/pull/5474

tatarize, 2021-05-14:
GUI Additions are a bit harder to do since I have to fire up wxGlide to do that work and there's some chance the camera doesn't have those options available. Part of the neat part of this request is that I'd merely have to hook that stuff up and control it with console. And if there was a need for a gui there I'd likely prefer a different window altogether to manipulate the options. Since I don't foresee this being super-important, it's just with the build in with OpenCV its likely just a few hours of work.

inspectionsbybob, 2021-05-14:
It only is important on fine placement work where a few MM make a
difference, or you are trying to line up a cut on an existing piece and
really need to see clearly a registration line or cut line...  If I am just
avoiding the last cutout by an inch or so, or missing my magnets (hold
downs) it isn't important. Where "tight registration" is called out, I want
the focus and brightness.

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2022-07-20 by tatarize