meerk40t/meerk40t #386: Request for Comment: Operations Upgrade / Corrections.

tatarize, 2021-03-10:
The operations need to overtly save without classification necessary. 

Workflows:
* If the svg has no operations
     1. Use dynamic previously set operations
     2.  Use static predefined saved operations
* If the svg has operations.
     1. Switch to dynamic operations and set operations to the defined loaded operation.
     2. Then, classify the elements into those operations.
* If the svg has has operations and overtly linked operation.
     1. Switch to dynamic operations and set those operations to the defined operation set.
     2. Then, only use the overtly linked operation values.

The operations values should allow loading and saving of static predefined operation which are firm. Meaning that if an operation does not match the given set these are not added to the set or modified. All modifications are potentially lost unless this is overtly saved. These values if set will be used by default and restored unless overtly saved by the user. If set to the 0th item they are purely dynamic and save whatever was used last-time.

 ---

Goals:
* We should allow predefined objects to be loaded easily and successfully without regard to the color operations links.
* We should permit highly functional speed and materials tests without regard to the objects being used. Save the objects and the elements within the object.
* We should be heavily backwards compatible with 0.6.x versions.
* We should save and restore command-operations.
* We should seek to remedy the distinct issue of operations becoming grossly malformed and causing new users to become confused as to how they should work.
* * We should seek to allow a makeshift materials database where users can predefine semi-static settings, and must overtly update them. Conforming to full CRUD interactions. Create, Reload, Update, Destroy.

inspectionsbybob, 2021-03-11:
In English?
Like "be compatible with 0.6..."
Like "allow a makeshift materials database..."
Not sure about the SVG files having operations in them... Inkscape, for example, doesn't understand raster/engrave/cut... it understands layers, objects, and colors...  If objects don't match colors or aren't a recognizable "image" dropping them into a default operation makes sense, as long as it is easy to edit.  The current operations menu has too much stuff in it, some of which is superfluous (like files??)

tatarize, 2021-03-12:
Files from the 0.7.0 modifications should at least load without issue in 0.6.x even if they don't classify things correct.

A materials database is a bunch of settings that are defined for different materials to give folks an idea of what settings should work for that material, or at least in the right ballpark. Then they adjust those settings, resave them and have closer settings for their machine.

The idea is to make the operations a lot easier. They currently can easily confuse folks and that's a problem.

inspectionsbybob, 2021-03-12:
For the material database, I know several people have developed their own
sample template that has a number of power levels on it.  I had one, and
now realize that it was too complicated with too many variables.

As part of the database I would suggest a power template... a -simple-
template with graduated outputs... don't need different size squares or
circles, just different powers so you could judge how much power is needed
to raster or cut a material.  actually maybe 3 identical templates for
hard/thick materials, medium materials, and light/thin materials, where the
delivered power to the material almost overlaps with the three templates.

I run 90% of my jobs at 6ma, and adjust power with speed, ppi and step, all
of which are divisors. I have run at multi-pass 9mm/sec-1000ppi  for cut
and 200mm/sec-500ppi step 2 on really hard dense stuff, to 20mm/sec-500ppi
for fabric.

i haven't tried cutting above 20mm/sec...

As i learn, I have found that i can't cut/engrave above certain speed
without thrashing the machine... even though it can raster at 400mm/sec, it
can't cut/engrave at anything close to that... even if I it is thin/light
material.

Anyway... a pre-done delivered power template...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

tatarize, 2021-03-12:
Part of the issue there is there's no good way to flag items as belonging in a particular operation. So they *must* classify by color, even if you put them specifically in the correct sections. And saved the file from MeerK40t. It can very easily load wrongly if you didn't give them each a different color. That's also a massively unneeded hoop to jump through. And this entire thing should be greatly simplified without denying power-users any fine grain control.

inspectionsbybob, 2021-03-12:
No Issue there, as Whisper also uses color...   If there was another
popular control software that did something different, that would be one
thing, but I don't think there is one.  Lightburn is expensive, and I don't
know how it classifies stuff coming in.  If you had NAMED LAYERS, that
would work, but I don't know how much extra work that would be, or who
would do it as that would be in Inkscape or whatever software prepared
the SVG.

Bob Sisson ACI, BVI

tatarize, 2021-03-12:
Lightburn duplicates the RDWorks version (which you can download for free) of doing this. There are set colors. These colors are specifically given an action like cut, or raster, or dot (engrave with dots). There is a color bar at the bottom of the screen and you can select some objects and click the color bar option and assign those elements to that color in a 1:1 fashion. You do not ever put the same thing in two different colors. And each color specific to one operation.

inspectionsbybob, 2021-03-12:
So, colors work...
Now worries about having colors...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2022-07-20 by tatarize