meerk40t/meerk40t #387: Error report

inspectionsbybob, 2021-03-10:
MeerK40t crash log. Version: 0.6.21
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wx\core.py", line 3407, in <lambda>
  File "Kernel.py", line 750, in process_queue
  File "wxMeerK40t.py", line 836, in on_rebuild_tree_signal
  File "wxMeerK40t.py", line 2114, in rebuild_tree
  File "wxMeerK40t.py", line 2143, in build_tree
  File "wxMeerK40t.py", line 2142, in build_tree
  File "wxMeerK40t.py", line 1870, in __init__
  File "svgelements.py", line 3560, in bbox
  File "svgelements.py", line 3561, in <listcomp>
  File "svgelements.py", line 5202, in bbox
  File "svgelements.py", line 5196, in angle_inv
ZeroDivisionError: float division by zero

tatarize, 2021-03-11:
Hm. That's a  pretty deep bug. I'll get it. Probably a svgelements bug though. So it's a lot more stuff to update.

tatarize, 2021-03-11:
`Path("M0,0A0,0 0 0 0 0,0z").bbox()` svgelements replication is a 0 sweep arc with the bounding box.

tatarize, 2021-03-11:
Raised the relevant issue in svgelements testing for additional fail conditions.

tatarize, 2021-03-11:
Not an issue in svgpathtools. I'm not sure I love the implementations there but it's reasonable enough.

tatarize, 2021-03-11:
https://github.com/meerk40t/meerk40t/commit/59f66b095a20a632de50ba5d6d66bd1c71d0057f
https://github.com/meerk40t/meerk40t/commit/a1b1ed3cc7b73299560daa11da79b7c74530f873
https://github.com/meerk40t/svgelements/pull/98/commits/cfa81e19e862a0a0dae65af3142cc531c4a0eb58

Fixes all the places it shows up. Degenerate arcs aren't that common.

inspectionsbybob, 2021-03-11:
I was trying out
https://en.makercase.com/#/basicbox
and asked for the pieces to touch to save travel time and cut a common
edge...
redownloaded without them touching and had no error.

ps...nice site for making project boxes...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2021-05-20 by tatarize